OctoPrint/OctoPrint #1925: [Regression] Webcam HTTP 401 doesn't trigger browser auth any more (shows error instead)

Depaulicious, 2017-05-18:
#### What did you expect to happen and what happened instead?

On `stable` the webcam wouldn't shop up if there were any issues. In `maintenance` it shows a user-friendly "not working" screen. However while in `stable` the fact that the image was loaded unconditionally triggered the browser authentication window if the stream was behind HTTP authentication (HTTP 401). In `maintenance` it doesn't any more.

Opening the webcam URL manually in a new tab and authenticating works around this.

#### Branch & Commit or Version of OctoPrint
`maintenance` @ 7770ee3d08389d

#### Operating System running OctoPrint

OctoPrint on Raspbian jessie

#### Browser and Version of Browser, Operating System running Browser

Google Chrome 58.0.3029.81 on Ubuntu 17.04

#### Link to octoprint.log

Does not apply.

#### Link to contents of terminal tab or serial.log

Does not apply.

#### Link to contents of Javascript console in the browser

https://gist.github.com/Depaulicious/1f6722c4f90cdeb9a19dc49d36bb0649

#### Screenshot(s) or video(s) showing the problem:

[image]

foosel, 2017-05-19:
@Depaulicious you didn't copy ALL of the template, and hence the bot couldn't detect that you in fact had provided all information. Marking it as complete manually.

Just as a quick question for reproduction: Your webcam's stream url is protected through basic auth, but only that, right? What I mean is, it's not like you have OctoPrint AND the cam protected by the same basic auth terminated on the same server, but instead it's a case of only the camera requiring credentials (or requiring different credentials than OctoPrint), correct?

Depaulicious, 2017-05-19:
Both the camera AND OctoPrint are protected, but when I'm home, I always just put the raspberry pi ip address so it doesn't have to go through the internet and back. In the first situation there's no problem (when I open OctoPrint I'm asked to authenticate and the whole third level domain it is bound to is unlocked), but in the second case, OctoPrint is opened though the lan (no Apache, no auth), but the webcam is still configured to use the external URL, so the issue occurs.

I didn't have time to check the code, which is why i opened the issue, but I believe it's just some JavaScript Ajax (or similar) call calls the error callback because the http status is 4**, though 401 should be ignored.

foosel, 2017-05-19:
It's actually an error handler attached to the `img` tag now that wasn't there before. No fancy stuff, just an `onerror` callback. I'll have to check why having that in place apparently suppresses showing authentication windows.

foosel, 2017-05-19:
Hm... I've tried to mirror your setup by setting up basic auth for a webcam in my LAN and embedding that, but (using a private browsing tab to make sure nothing interferes with reproduction) I don't get a authentication prompt on master/1.3.2 stable either, just directly a 401 and a broken image icon. I'm... confused. You are absolutely positive that this does trigger an authentication prompt on 1.3.2 with your setup, in a fresh browser session?

Depaulicious, 2017-05-19:
@foosel omg sorry it was a long day 😂 It does _not_ trigger authentication, but it still does show the "empty" image, so I would right click it, open it in a new tab and authenticate.
It could be handled better anyway, for example when 401 occurs show a message like "The webcam requires HTTP authentication. Click here to open it in a new tab and authenticate".

foosel, 2017-05-19:
Argh! And I just started to doubt my sanity :D Ok, now that does make sense. Thing is, I can't detect that 401 at all, I only see an "error" type Event, not what kind of response triggered that error. As I said, not doing an AJAX call here, just have a handler attached to the `load` and `error` events on the image and handling those (disabling "loading..." on load, hiding image and showing "oops" on error). 

Hiding the (broken) image was actually intentional ("It's not a bug, it's a feature!") because in the past I got a lot of tickets about that broken image from confused users who didn't know what to do with that. And keeping it visible when there already is a an error message explaining that stuff couldn't be loaded is redundant and also looks broken.

What I could do though would be adding something like "The configured stream URL is:  `...`", with the URL as a link: 

[image]

That would help you, provide some more info to the user and still fit the above considerations. How does that sound?

foosel, 2017-05-19:
Forgot to reference this ticket, the aboce is implemented in 0ad09e422d4aac865cbcd991b7998dfa7bc8e9ee

Depaulicious, 2017-05-19:
Very good :smile: Maybe the string should be changed to "It might not be correctly configured, or require authentication. [...]"

So that it's clearer to new users.

foosel, 2017-05-19:
[x] done :smile:

Logged in as admin:

[image]

Logged in as user:

[image]

Not logged in:

[image]

Depaulicious, 2017-05-19:
That was fast 😂 Tested and works well :)

closed 2017-09-24 by Depaulicious