OctoPrint/OctoPrint #1926: C270 Webcam

abreumateus, 2017-05-18:
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
that are linked ^-- just up there in the big yellow box. Also read
the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ..."), that belongs on
the mailing list or the G+ community (both linked in the "guidelines
for contributing" linked above, read it!), NOT here.

Mark requests with a "[Request]" prefix in the title please. Fully fill
out the bug reporting template for bug reports (if you don't know where
to find some information - it's all described in the contribution
guidelines linked up there in the big yellow box).

When reporting a bug do NOT delete any lines from the template but
those enclosed in [ and ] - and those please DO delete, they are
only provided for your information and removing them makes your
ticket more readable :)

Make sure any bug you want to repor tis still present with the CURRENT
OctoPrint version and that it does not vanish when you start OctoPrint
in safe mode - how to do that is also explained in the contribution
guidelines linked up there in the big yellow box.

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

#### What were you doing?

[Please describe the steps to reproduce your issue. Be as specific as
possible here. The maintainers will need to reproduce your issue in
order to fix it and that is not possible if they don't know what you
did to get it to happen in the first place.

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen and what happened instead?

#### Branch & Commit or Version of OctoPrint

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Operating System running OctoPrint

[OctoPi, Linux, Windows, MacOS, something else? With version please.]

#### Printer model & used firmware incl. version

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser

[If applicable, always include if unsure or reporting UI issues.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting any kind of communication issues between OctoPrint and your printer.
Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) or video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

ntoff, 2017-05-19:
>Octopi

Wrong bug tracker, octopi is over here: https://github.com/guysoft/OctoPi/

Also you don't need to manually install mjpeg streamer for octopi, everything is already installed and set up for you. That you did those manual steps probably broke something, wipe the SD card, flash the octopi image again and check here: https://github.com/foosel/OctoPrint/wiki/Webcams-known-to-work for your webcam and see if it requires any extra command line parameters to function correctly. Enter those parameters in /boot/octopi.txt (or if you're accessing the sd card via windows and an sd reader, it's in the only partition you'll find on the card in octopi.txt)

foosel, 2017-05-19:
What @ntoff said. See also [this FAQ entry](https://github.com/foosel/OctoPrint/wiki/FAQ#my-webcam-is-not-working).

Closing because not an issue with OctoPrint, more a case of "I need help with" -> get in touch on the mailing list or the G+ community please.

As a side note, I run the C270 on several Pi's with OctoPi and it works without a flaw out of the box without any additional steps being necessary.

closed 2017-05-19 by foosel

abreumateus, 2017-05-19:
Do not close untill i solve my problem.

I'm using an Pi Zero, maybe it's related to the type of my Pi?

foosel, 2017-05-19:
This is the bug tracker for a piece of software called OctoPrint, not a support forum. Your problem is not a bug with said software but rather an issue with your particular setup (Pi Zero + OctoPi + a Logitech (?) C270) that you are requesting help with. As repeatedly stated, those kind of problems are better suited for the existing support channels, as in, the mailing list or the G+ community, because there are WAY more people there that might be able to help you troubleshoot your problem.

This is nothing against you, it's simply to keep things manageable - maintaining this piece of software is a lot of work, and I can't do that work if at the same time I also need to field support requests from the thousands of users world wide on top of everything else. It simply does not scale. Hence the redirect of such requests towards the user community, in the hopes that people will help each other there (which usually appears to work nicely).

bigten055, 2019-10-25:
This camera worked good for 2 weeks on python with open cv4 on pi 4 then all of a sudden it started giving me these errors. I have an image saved from when it was working fine so i reverted back and is still giving me errors. weird thing is camera still works on pi 3B+ on cv4 i can only assume that it is hardware related.