OctoPrint/OctoPrint #1927: Webcam not disabled if browser tab closed

aviatorken89, 2017-05-19:
**What were you doing?**

Closing the browser tab for OctoPrint in Safari and Firefox.

**What did you expect to happen and what happened instead?**

I expected the webcam to be disabled when the browser tab for OctoPrint is closed, but it was still enabled. I was able to test and confirm this because I also have a GPIO pin as high whenever the webcam is active, and low when the webcam is disabled.

**Branch & Commit or Version of OctoPrint**

Devel

**Operating System running OctoPrint**

OctoPi

**Browser and Version of Browser, Operating System running Browser**

Safari 10.1.1 on macOS 10.12.5 and Firefox 53.0.2 on macOS 10.12.5

I have read the FAQ.

I have fixed this issue in this commit: [](https://github.com/aviatorken89/OctoPrint/commit/f32bd1972389c391b1132b36e9cb244437cbf08c)

foosel, 2017-05-19:
I tested in both Safari and Firefox on macOS Sierra (and also on my local Firefox and Chrome under Win10). The webcam stream stopped streaming just fine on tab close - and if it didn't when the tab unloads that would definitely sound like a browser bug.

I don't know what kind of webcam you have running, or how you've wired it up, but I'm definitely not seeing any kind of continued HTTP traffic from the camera to the browser when the tab is closed. OctoPrint merely embeds the camera stream, so it doesn't have any control over enabling or disabling it in any way on the hardware layer (and if that's different for you, it's some feature of your particular setup), but unloading/closing the tab definitely closes any existing connections to the server here.

Could you please test with `tcpdump` or a similar tool and verify you are indeed seeing *traffic* from the camera to your closed browser?

aviatorken89, 2017-05-19:
Maybe I should rephrase, the disable_webcam method is not called on browser tab close. With the fix above it works because the GPIO pin I'm using goes to 0.0 volts.

foosel, 2017-05-19:
Why should it be called? The browser tab is closing, everything in the browser tab is about to be thrown away anyhow, there's no need to do cleanup that the browser should usually do anyhow.

I don't understand why your GPIO pin gets influenced by the presence of this function call, but in general, there really should be no difference at all with regards to the webcam stream unloading along side the rest of the tab contents and hence no need for this call. Especially since all the call does is setting the `src` attribute of the webcam stream image to an empty string, and it doesn't even do THAT on Safari due to a Safari bug, so you shouldn't even see a difference there at all.

I'd like to understand what's going on on your end there before adding usually unnecessary calls and event handlers, that's all.

aviatorken89, 2017-05-19:
What I have setup is an LED light mounted on the webcam that turns on and off for whenever the webcam is called to be enabled or disabled respectively, so even if the webcam stream is torn down on the internal cleanup, the pin still remains high and the LED light is still on because _disableWebcam() wasn't called. Sorry, I should've stated more originally :)

foosel, 2017-05-19:
Ah, so you've hooked into the `_disableWebcam` function somehow and are toggling said GPIO pin? Is there some minimal example of how you've set this up software wise so I could get a clear picture? :)

From my current understanding though, this sounds more like a request to also call this function on tab close, rather than a bug report, or am I still missing something here?

aviatorken89, 2017-05-19:
Yep, pretty much exactly that. Although it took me a few days to figure out how to implement it I'm not sure how to have a minimal example. I had to add my own commands in the OctoPrint printer interface api basically; I think it was split across 4-5 files to get working. I didn't see a `onTabClose` function otherwise I would've put it there :)

foosel, 2017-05-19:
I took a look at your fork... First of all, if I'm not mistaken the way you are doing things here will produce interesting results if more than one client connects to the web interface. E.g. Client 1 and Client 2 connect. Led gets turned on (twice). Client 2 disconnects. Led gets turned off. Hilarity ensues. I'm also not sure why you modeled the "someone is watching the camera" into the printer interface, which has absolutely nothing to do with the camera.

Then, I'd strongly recommend against modifying the sources directly and instead do your changes through one or more customized plugins. The way you are doing it now you'll run in merge errors all the time, and it would also allow you to e.g. call whatever additional calls you need to do on unload of the browser without any hackish bits of code in places where they don't really belong from an architectural point of view.

Adding your commit doesn't make sense for core OctoPrint, since all it does (and all it ever will do) in the `_disableWebcam` function is disabling the webcam *in the browser* by setting the URL to an empty string (well, unless you are running this in Safari, because Safari is stupid and will start duplicating streams if we do that). This is a purely client-side task, only in place to save bandwidth when nobody is watching the webcam, and there should never be any interaction with the server going on here.

aviatorken89, 2017-05-19:
I wanted to make a plugin, but had no luck trying to figure out how to bundle it into the OctoPrint build itself and couldn't find any documentation for putting it directly in the build so that when I build with OctoPi it's already there on the sd card, so I had to resort to modifying the other source files. Just a temporary solution until I can acquire the knowledge to build the plugin right in, as that's what I want.

foosel, 2017-05-19:
Two options.

  * Modify the OctoPi build script or rather `chroot_script` specifically (or add an OctoPi variant, even better) to do a `/home/pi/oprint/bin/pip install <yourPluginsSourceArchiveOnGithub>` as part of the build. Done. Cleanest solution. Example for an OctoPi variant that also installs a custom OctoPrint plugin see [here](https://github.com/foosel/OctoPi/tree/newdevel/src/variants/netconnectd).
  * Put your plugin into `src/octoprint/plugins` of your OctoPrint fork. That bundles it. Done. Not the best solution, since it still makes your OctoPrint version diverge from mainline, meaning it will be unsupported - but that's your choice. Merge conflicts should be pretty much non-existent since your plugin will live in its own folder.

If you don't mind, I'm closing this now though, since as mentioned above what you are requesting OctoPrint to do here makes only sense for your specific modified version and I'd rather not add otherwise unnecessary calls to support such things. Hope you understand.

closed 2017-05-19 by foosel

aviatorken89, 2017-05-19:
I understand how to put the plugin into the build, but I wouldn't know how to call my plugin from the _enableWebcam and _disableWebcam functions at all. Thanks :)

foosel, 2017-05-19:
Somewhat hack-ish but workable: Declare a viewmodel (doesn't need to bound to anything), declare the `ControlViewModel` as a dependency, wrap `_enableWebcam` and `_disableWebcam`, e.g. (untested)

``` javascript
$(function() {
  function MyFancyViewModel(parameters) {
    var self = this;
    self.controlViewModel = parameters[0];

    // ...

    self.onStartup = function() {
      var originalEnableWebcam = self.controlViewModel._enableWebcam;
      self.controlViewModel._enableWebcam = function() {
        // do whatever you need to do before...
        originalEnableWebcam.apply(self, arguments);
        // do whatever you need to do after...
      }

      // likewise for _disableWebcam
    }
  }

  OCTOPRINT_VIEWMODELS.push([
      MyFancyViewModel,
      ["controlViewModel"],
      []
  ]);
}
```

The same view model should also hold your client functions for your LED light, which should be provided by your plugin's API (e.g. via `SimpleApiPlugin`) and not somewhat patched into system components they have nothing to do with like the printer interface.

nophead, 2017-05-21:
What if you have the camera view open on several tabs, or on different
client machines? The camera light should only go out when the last one is
closed. Seems like mjpg-streamer should responsible for disabling the
camera when it has no connections.

aviatorken89, 2017-05-21:
@nophead Why would this matter if I'm the only one using my printer?

nophead, 2017-05-21:
Well I am the only one operating my printers but I often have OctoPrint open on several computers.

aviatorken89, 2017-05-22:
I made a plugin and it looks like it installs fine, but octoprint isn't detecting it so it doesn't work :( Can you look at my plugin repo? @foosel https://github.com/aviatorken89/OctoPrint-WebcamLED

Thank you!

aviatorken89, 2017-05-23:
Any help would be appreciated

foosel, 2017-05-24:
@aviatorken89 sorry, but I really don't have the time right now. Try the mailinglist please, there are more people that might be able to help there.