OctoPrint/OctoPrint #1928: Problem with Temp Chart

KC703, 2017-05-20:
#### What were you doing?
While normal printing the temperature graph is not displaying accurate temps.  

The issue may be related to GCODE Scripts within Octoprint, I use a "Start script" to preheat my bed prior to heating the hotend:
M190 S85
Basically, when the bed reaches 85C, start the print which heats the hotend to 245C and bed to 110C.

#### What did you expect to happen and what happened instead?
The temp graph to display the temp of the bed and hotend.

#### Branch & Commit or Version of OctoPrint
OctoPrint 1.4.0.dev368+g3e5d82c (devel branch

#### Operating System running OctoPrint
Raspberry Pi (Octopi)

#### Printer model & used firmware incl. version
Lulzbot TAZ 5, Marlin 1.0.0

#### Browser and Version of Browser, Operating System running Browser
Windows 10, Chrome  Version 58.0.3029.110 (64-bit)

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.]

#### Link to contents of terminal tab or serial.log
https://www.dropbox.com/s/0d7pk08ocf72k92/serial.log?dl=0

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) or video(s) showing the problem:
The following picture is the temp graph at the end of my GCODE Start script:
[image]

During Printing:
[image]

Later in the Print:
[image]

A picture of the LCD Screen of the printer showing correct temps:
[image]

Yes, I have read the FAQ.

KC703, 2017-05-21:
I forgot to add that this issue started in the recent two commits (~5/17)

foosel, 2017-05-22:
Hm, I cannot reproduce this. Testing with the exact same version, currently printing and happily reporting the (correct) temperature.

What I notice from your provided `serial.log` is that your firmware decides it's done heating your hotend to 245°C when it has only reached 235°C:

```
2017-05-20 14:00:43,611 - Recv: T:225.4 E:0 W:?
2017-05-20 14:00:44,611 - Recv: T:226.6 E:0 W:?
2017-05-20 14:00:45,612 - Recv: T:228.1 E:0 W:?
2017-05-20 14:00:46,613 - Recv: T:229.1 E:0 W:?
2017-05-20 14:00:47,619 - Recv: T:230.2 E:0 W:?
2017-05-20 14:00:48,619 - Recv: T:231.6 E:0 W:?
2017-05-20 14:00:49,664 - Recv: T:232.9 E:0 W:?
2017-05-20 14:00:50,663 - Recv: T:234.1 E:0 W:?
2017-05-20 14:00:51,666 - Recv: T:235.5 E:0 W:0
2017-05-20 14:00:51,851 - Recv: ok
```

And the reason your temperature graph then doesn't update is that for some reason OctoPrint apparently is not sending out `M105` queries at all to retrieve the temperature. The `octoprint.log` might give some hints as to why - please also provide that, it's missing up there. Also try to reproduce the problem in safe mode please, to rule out any plugins from interfering here.

nophead, 2017-05-22:
Looks like the wait time is set to 0 which is not the best idea. Just because the heater block has got to temperature (or not in this case) it takes some time for the filament to catch up and for some hot ends to expand to the normal length.

KC703, 2017-05-23:
Thanks for looking into this.  I did a little more testing, and unloaded a
few plugins which I wasn't using.  Unloading the "Tempsgraph" and the
regular Octoprint graph is working.

How do you enable a disabled plugin... the radio button disappears, leaving
just the trashcan to delete the plugin.

Regarding the "0" wait time, I usually print a skirt or brim to ensure the
hotend is primed...  :-)

Safe to close this issue.  Thanks!

foosel, 2017-05-23:
@KC703 huh... I thought that plugin was frontend only, which wouldn't explain why no `M105` is sent (that's a quite crucial part of the communication layer deep down in the backend). 

Happy to hear that disabling the plugin resolved it for you, but I'm wondering *why* ;)

Re enabling disabled plugins:

[image]

But enabling plugins that are not bundled is disabled in safe mode, so if you are running that, that would explain it.

Closing since apparently the problem was caused by a plugin (even though I don't understand that ;))

closed 2017-05-23 by foosel

KC703, 2017-05-23:
Hmm... I don't think I'm in "Safe Mode".  Here's what I see when trying to
re-enable plugins.
[image: Inline image 1]

foosel, 2017-05-24:
@KC703 can you go to the Github issue and post the image there instead of replying to the mail? The image didn't make it through.

KC703, 2017-05-25:
Sorry.  Here's what I see when trying to re-enable plugins: 
[image]

The radio button is missing.

foosel, 2017-05-29:
@KC703 That turned out to be a bug introduced by #1915, because it was referring to a non existing icon class. That's been fixed with the above commit (which should be present on the `maintenance` and `devel` branches)

KC703, 2017-05-29:
Hmm... I'm running:
OctoPrint 1.4.0.dev387+g905c05b (devel branch)

It looks to still be present?

KC703, 2017-05-29:
Sorry that last message may have been premature. 

Just received an update to OctoPrint 1.4.0.dev389+g528bbba (devel branch) and the radio button to enable/disable the plugin is back.

Thanks!
Ken

foosel, 2017-05-30:
👍