OctoPrint/OctoPrint #1930: Website visitors can go to /webcam/control.htm and modify webcam settings with no authentication

OmgItsBkid, 2017-05-20:
#### What were you doing?

Browse directly to http://yoursite.com/webcam/control.htm and you will be presented with webcam controls that work in real time and affect all users viewing the webcam. This can be done without the person logging into OctoPrint.

#### What did you expect to happen and what happened instead?

I did not expect this page to be available (as it does not appear to be linked from what I can tell), or at least available behind the OP login.

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.2 (master branch)

#### Operating System running OctoPrint

OctoPi 0.13

#### Printer model & used firmware incl. version

N/A

#### Browser and Version of Browser, Operating System running Browser

N/A really, but Version 58.0.3029.96 (64-bit) on Windows, and Chrome 58.0.3029.83 on Android

#### Link to octoprint.log

https://pastebin.com/qU0kTKSj

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) or video(s) showing the problem:

[image]

I have read the FAQ. ;)

ntoff, 2017-05-21:
>(as it does not appear to be linked from what I can tell), or at least available behind the OP login.

Because it's not a part of octoprint, it's part of mjpeg streamer which comes bundled in the octo*pi* distribution, so it's not something octo*print* has any control over.

Bugs for octopi are tracked here: https://github.com/guysoft/OctoPi/ and may be worth bringing up so it can be removed or somehow hidden behind octopi's bundled haproxy redirection to only allow localhost access.

OmgItsBkid, 2017-05-21:
I thought it might be something for OctoPi possibly, but I wasn't 100% on it as I haven't installed OctoPrint manually, and didn't know what was and was not included.

At the very least, this can warn some users on this side, and I'll post it over on the OctoPi side as well.

guysoft, 2017-05-22:
@foosel Suggest we close this here and continue in OctoPi issue tracker.

foosel, 2017-05-22:
@guysoft Agreed.

closed 2017-05-22 by foosel