OctoPrint/OctoPrint #1934: Timelapse capture doesn't start anymore after the first print

gege2b, 2017-05-24:
Hi there

#### What were you doing?

Doing many prints in a row, with mjpg server, webcam & timelapse enabled, but timelapse capture start only for the first print

The control tab shows the stream as expected

Prints could be of the same or differents objects, it makes no difference

#### What did you expect to happen and what happened instead?

Timelapse to start at every print.
But instead, no capture at all after the first print and successfull timelapse (the print is going fine, trough)

If I do a print just after a server restart, the timelapse is captured normally

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.4.0.dev368+g3e5d82c (devel branch)

#### Operating System running OctoPrint

current raspbian lite on Raspberry pi 3, 
webcam is a "Trust HD" with no more info on it :) 

#### Printer model & used firmware incl. version

Not relevant

#### Browser and Version of Browser, Operating System running Browser

Windows 10, Firefox 53 x64

#### Link to octoprint.log

[Octoprint log](https://framabin.org/?3798f3e57182aef8#lWzsjspsQYF4dT6uuTApdjCgsQmje3jTHFDt2/FjoAM=)

(but I see nothing about timelapse in it)
Please tell me if you want me to supply different info

#### Link to contents of terminal tab or serial.log

not relevant

#### Link to contents of Javascript console in the browser

not relevant

#### Screenshot(s) or video(s) showing the problem:

not relevant

I have read the FAQ.

foosel, 2017-05-24:
Thanks for reporting this, that was also present in 1.3.3rc2 (well, actually also earlier than that, but covered up by a setting that got changed in 1.3.3rc1), so got caught just in time.

Already fixed on `maintenance`, soon `devel`, will go into 1.3.3rc3 I hope to release later today.

foosel, 2017-05-24:
Small correction, already fixed on `staging/maintenance`, soon on `maintenance` and `devel`

commit 2a600d7d (2017-05-24): Fix consecutive timed timelapses without postroll capture

Due to the timer never properly resetting to None only one timelapse
was ever done.

Introduced in 7e2382fed4199ee50db14e40cc8b03c694894a04, covered until
315a80a7a7f7336747937e1b17b9706e77859cff.

Closes #1934
closed 2017-05-31 by foosel