OctoPrint/OctoPrint #1935: [RC Feedback] Feedback on the 1.3.3rc3 Release Candidate

foosel, 2017-05-24:
Please provide *general* feedback on your experience with the 1.3.3rc3 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

If you run into any obvious bugs, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).

ctgreybeard, 2017-05-24:
I'm not sure this is a regression as I think I have seen the connection issue in 1.3.3RC2 before but I will report here so I have posted it at least.

Upon first startup of OctoPrint, after having disconnected from the printer before restarting, I get a connection failure the first time I attempt to connect. My port is set to AUTO, baud to 115200. Upon the second connect, without changing anything the connection succeeds. Here is the terminal window:

```
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Error while connecting to /dev/ttyACM0: 'Timeout'
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
Connected to: Serial<id=0x74ad2390, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: start
Send: N0 M110 N0*125
Recv: echo:echo: Last Updated: Feb 14 2017 15:35:19 | Author: (none, default config)
Recv: Compiled: Feb 14 2017
Recv: echo: Free Memory: 2670  PlannerBufferBytes: 1280
Recv: echo:Hardcoded Default Settings Loaded
Recv: echo:SD card ok
Recv: Invalid M code.
Recv: ok
Changing monitoring state from 'Connecting' to 'Operational'
Send: N0 M110 N0*125
Recv: Invalid M code.
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin V1.0.2; Sprinter/grbl mashup for gen6 FIRMWARE_URL:https://github.com/prusa3d/Prusa-i3-Plus/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK2 EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Recv: ok
Send: M117 Hello!
Recv: ok
Send: M115
Recv: FIRMWARE_NAME:Marlin V1.0.2; Sprinter/grbl mashup for gen6 FIRMWARE_URL:https://github.com/prusa3d/Prusa-i3-Plus/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK2 EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Recv: ok
```

Note the `Timeout` in line 4. This was after the upgrade to RC3. I had intentionally disconnected using the Disconnect button prior to performing the upgrade.

I also, as an aside, noticed that there are two instances of:
```
Send: N0 M110 N0*125
Recv: Invalid M code.
```

I'm not familiar enough with the GCODE nuances to know whether that is really an error or something that the Prusa firmware invents.

foosel, 2017-05-24:
That `Invalid M code` message actually sounds like what the firmware read is not actually the `N0 M110 N0*125` that was sent to it. I'm pretty sure that they do support `M110`. Something there tickles my memory though (firmware related, not OctoPrint related), I hope I'll remember what it was. Certainly looks like something hiccuping on the serial though.

The `Timeout` can only come from an attempted port auto-detect, so my guess is with your particular board that can take a bit too long to succeed. This is also a bit of code that was last changed in early 2016, so I'm fairly sure it's not a regression from 1.3.2. It's still good to report things like that 👍

ctgreybeard, 2017-05-24:
How long is too long? The message appears immediately, not even one second later. If the timeout is in seconds then it isn't waiting at all. The hardware, BTW, is a Pi3 if that helps. Would a serial log help debug it? It's a minor issue if you don't feel the need to pursue it.

foosel, 2017-05-24:
I just took another look into that bit of auto-detection code that's the source of the `Timeout` message and apparently that particular error message gets triggered when entering programming mode fails due to a read timeout on the serial line (with a read timeout of a micro second as far as I can see O_O). Trying to enter programming mode in order to determine if it's a printer or not is still a relic from the Cura-Fork days, and something I've been meaning to get rid of for a while. Definitely not something that was changed in the past couple of versions. Still, I guess it's time to take a closer look at the "getting rid off" part. For now, I can't do much about it besides recommend to just not use auto detection when you only have one serial port anyhow. I'll see what I can do concerning this particular annoyance in 1.3.4.

dookie2000ca, 2017-05-24:
Smooth upgrade from rc2 to rc3 on my OctoPi 0.14rc1 installation.  
- Printer: UltiBots D300VS 
- Firmware: RepRapFirmware for Duet WiFi 1.15e

ctgreybeard, 2017-05-24:
I wouldn't put any priority at all on the issue of the timeout. For me it occurs infrequently because the printers is normally connected all the time.

milindur, 2017-05-25:
Upgraded from rc2 to rc3, so far no issues, but I am still fairly new to 3D printing and OctoPrint.

Fiiti, 2017-05-26:
**Referring to your last video and the missing feedbacks:**
Easy upgrade fom rc2 to 1.3.3rc3. Everything is working very well, except a little issue with the feature for the size check. I am going to open an issue in the other place.
I love this stuff. Very good work!
Many thanks for OctoPrint!
Using Cura 2.5 Beta3 with OctoPrint PlugIn.
Prusa i3 Clone with Marlin 1.0.
Logitech C270 Cam
Raspi3b.

Best regards,
Fiiti

xylo04, 2017-05-27:
No issues to report using 1.3.3rc3! Thanks as always.

JosvdP, 2017-05-28:
I'm not sure if this is an obvious bug or a mistake on my end, so I'll post here first.

I run 1.3.2 and disabled almost all plugins prior to the update. However, although the update appears to install succesfully, I'm returned in 1.3.2 nevertheless. Are there any update logs I can check? This is the general octoprint.log:

```
2017-05-28 10:04:16,396 - octoprint.plugins.softwareupdate - INFO - Starting update of octoprint to 1.3.3rc3...
2017-05-28 10:04:37,858 - octoprint.plugins.softwareupdate - INFO - Update of octoprint to 1.3.3rc3 successful!
2017-05-28 10:04:38,302 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2017-05-28 10:04:38,304 - octoprint.plugins.softwareupdate - INFO - Restarting...
2017-05-28 10:05:11,195 - octoprint.server - INFO - ******************************************************************************
2017-05-28 10:05:11,196 - octoprint.server - INFO - Starting OctoPrint 1.3.2 (master branch)
```

brianhanifin, 2017-05-28:
*Octopi 1.3.3rc3*
The Cost Plugin stopped displaying an amount one or two Release Candidates ago. The only reference I can find to it in octoprint.log is when it loads. I don't see any errors.

`|  Cost Plugin (0.1.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_cost
`

I have also [created an issue](https://github.com/jasiek/OctoPrint-Cost/issues/8) on the Cost plugin github.

foosel, 2017-05-29:
**To all of you**, huge thanks for the feedback, this is definitely way better than the usual silence after I put out an RC :) Glad to hear that apparently it so far looks like smooth sailing too!

@brianhanifin I cannot reproduce this issue with the Cost Plugin you are seeing. 1.3.3rc3, freshly uploaded and analysed file:

[image]

Same plugin version too. Anything in the JS error console maybe? Could be some weird conflict with another plugin.

brianhanifin, 2017-05-29:
I am starting to think I only noticed it when I was printing something small enough the price wasn't over 1 cent. I am going to close this. Please close this issue.

foosel, 2017-05-29:
@brianhanifin not closing this issue here just yet, since it's the general feedback ticket for 1.3.3rc3 :P but thanks for the update! :)

foosel, 2017-05-31:
Thanks everyone for the feedback, 1.3.3 was just released :)

closed 2017-05-31 by foosel