OctoPrint/OctoPrint #1936: [Brainstorming] PAM authentication

Petraea, 2017-05-26:
#### What were you doing?

All authentication for OctoPrint occurs using local password hashes. In addition to doing this, it would be good to be able to pass this safely off to PAM to allow users to authenticate using their own local passwords, and/or anything that PAM supports.

#### Branch & Commit or Version of OctoPrint

1.3.2

#### Operating System running OctoPrint

Linux.

foosel, 2017-05-30:
I think that goes a bit into the direction of #594 or in general making it easier to plug in additional authentication engines. See also [this comment](https://github.com/foosel/OctoPrint/issues/594#issuecomment-304923641) I just added to that ticket, explaining how something like that could already be achieved, just with a bit manual configuration and code instead of something fancy like a plugin.

Petraea, 2017-05-31:
A plugin system would be the most core-code-friendly system for this, for certain, but I agree, it's not too hard to construct a new UserManager and use that. I have a prototype piece of untested code at https://github.com/Petraea/OctoPrint/blob/master/src/octoprint/users.py#L405 that demonstrates this via the use of pwauth.

I was thinking of doing a merge request with this (once it's tested and bumped to devel), or would it be better to focus on an auth plugin system to wrapper this to allow for future hackery?

foosel, 2017-06-02:
Sorry for the delay, 1.3.3 and 1.3.4 kept me busy. 

PAM is very OS specific and OctoPrint in general is platform agnostic. So a PR for adding this would not be that great of an idea sadly, instead either maintaining the adjusted user manager separately with install instructions (= reconfiguration through `config.yaml` to use a separate class) or even better yet adding a plugin hook should be the way to go I think.

What I'm wondering about is the best approach to make something like this extendable by plugins in the best way. I mean, it probably will still require user intervention for configuration (you have 10 plugins providing authentication mechanisms, you need to choose which one(s) to use), unless one goes the route of some kind of chained authentication mechanism (run any plugins, if non match fall back to file based user manager). So maybe a simple [hook](http://docs.octoprint.org/en/master/plugins/hooks.html) to provide additional user manager sub classes, and a chained user manager that delegates to them?

jneilliii, 2017-06-07:
One way to accomplish this would be disabling octoprint's built in user administration and control through configuring haproxy's config.  I do mine that way with cert auth on top of an haproxy user set-up in the config and it works pretty well.

gillg, 2017-11-26:
I'm currently developing a plugin for LDAP. It works correctly, maybe I could inprove it...
See : https://github.com/gillg/OctoPrint-LDAP and don't hesitate give me your feedbacks