OctoPrint/OctoPrint #1937: Printer name in Big Letters and auto refresh on click...

derfred33, 2017-05-27:
Hi,

Just wanted to leave a couple of simple to implement suggestions, one is the printer name because when you have several instances on the same browser its useful to know what is the printer you´re viewing.

Another simple request was to add a simple javascript to refresh the camera image iframe or img when the Control tab is clicked.

And btw, multiple file upload would be great too.

Thanks for the great app.

ntoff, 2017-05-28:
You can add whatever name you want, go to the appearance settings section and enter a page title, it shows up in the top navbar.

[image]

KevMags, 2017-05-28:
You can also change the host name with raspi-config so that the name shows in the address bar.

ntoff, 2017-05-29:
OP said several instances so I assumed they meant several instances of octoprint running on the same machine (so they'd all have the same hostname but different ports).

KevMags, 2017-05-29:
I assumed "instances on the same browser" to mean that OP uses multiple Pis therefore multiple IP addresses.  Maybe I misunderstood.

derfred33, 2017-05-29:
Thank you guys. Next time I´ll read all the settings menus :) I meant several instances on the same browser.

foosel, 2017-05-29:
> [printer name]

As stated above, already available ;)

> [refresh camera when switching to control tab]

It already does that because the camera stream is also unloaded if the control tab is left for more than 5 seconds. That behaviour is disabled in Safari though because that has a bug that leads to multiplying the stream on unload/reload, see #1780.

> [multiple file upload]

There's a PR open for that (see #1868). However that still has some issues to resolve before it can be merged.

Any objections to closing this, considering that what you are asking for is either already implemented or on the way there? ;)

ntoff, 2017-05-30:
>left for more than 5 seconds  

Can we maybe have that configurable? Being as there's no way to enter gcode from the control tab (yes, I know there are plugins), frequently going back and forth between "control" and "terminal", or "control" and "temperature" means the camera gets loaded and unloaded a whole bunch.

foosel, 2017-05-30:
^- we can, starting with 1.3.4