OctoPrint/OctoPrint #1939: Dual-Extruder printing bug

Qixuga, 2017-05-30:
#### What were you doing?

Complex Part with Support.
Dual-Extruder print, left (1) with PVA Support, right (0) with PLA(-HT).
Pre-Edit in Netfabb, sliced in Simplify3D.
64MB Gcode-File (all Files can be provided seperatly but not be public posted).

S3D used a "cool down" script to every extruder change.
May be Octoprint or the printer (controller/software) seems to hang up during second extruder change.
Always the system wait during reach the given temperature (which will be correct) but do not reache it and seems to be in a loop heating nearby the target temperature and never reach them - fall down and again.
a stop/break is possible, but the printer don't leave the look - only a octoprint-reboot (reset ?) help.

#### What did you expect to happen and what happened instead?

printing ;-)

#### Branch & Commit or Version of OctoPrint

actualy stable, 1.3.2

#### Operating System running OctoPrint

Raspberry Pi 2 with Rasbian.

#### Printer model & used firmware incl. version

Multec Multirap m420 (updated since last bug).

#### Browser and Version of Browser, Operating System running Browser

Chrome derivative (SRWiron), newest.

#### Link to octoprint.log

octoprint.log: https://gist.github.com/Qixuga/4e1236bd02484217eadd695599ca8969

#### Link to contents of terminal tab or serial.log

serial.log: https://gist.github.com/Qixuga/6974242f3fb4f68824733fcefa9fcbc0

#### Link to contents of Javascript console in the browser

-

#### Screenshot(s) or video(s) showing the problem:

- (.stl, .factory and .gcode File available, but not public)

I have read the FAQ.

Thank you!

foosel, 2017-05-30:
That is not OctoPrint, that is your printer stuck in that heat up loop:

```
2017-05-30 08:04:14,699 - Send: N25360 M109 S205 T0*41
2017-05-30 08:04:15,705 - Recv: T:165.2 E:0 W:?
2017-05-30 08:04:16,708 - Recv: T:165.3 E:0 W:?
2017-05-30 08:04:17,710 - Recv: T:165.4 E:0 W:?
2017-05-30 08:04:18,712 - Recv: T:165.5 E:0 W:?
2017-05-30 08:04:19,714 - Recv: T:165.6 E:0 W:?
2017-05-30 08:04:20,717 - Recv: T:165.8 E:0 W:?
2017-05-30 08:04:21,719 - Recv: T:166.0 E:0 W:?
2017-05-30 08:04:22,722 - Recv: T:166.2 E:0 W:?
2017-05-30 08:04:23,724 - Recv: T:166.4 E:0 W:?
2017-05-30 08:04:24,727 - Recv: T:166.7 E:0 W:?
2017-05-30 08:04:25,729 - Recv: T:167.2 E:0 W:?
2017-05-30 08:04:26,731 - Recv: T:167.5 E:0 W:?
2017-05-30 08:04:27,733 - Recv: T:168.1 E:0 W:?
2017-05-30 08:04:28,735 - Recv: T:168.5 E:0 W:?
2017-05-30 08:04:29,737 - Recv: T:169.0 E:0 W:?
2017-05-30 08:04:30,738 - Recv: T:169.5 E:0 W:?
2017-05-30 08:04:31,741 - Recv: T:170.1 E:0 W:?
2017-05-30 08:04:32,743 - Recv: T:170.6 E:0 W:?
2017-05-30 08:04:33,745 - Recv: T:171.3 E:0 W:?
2017-05-30 08:04:34,747 - Recv: T:172.0 E:0 W:?
2017-05-30 08:04:35,749 - Recv: T:172.7 E:0 W:?
2017-05-30 08:04:36,751 - Recv: T:173.3 E:0 W:?
2017-05-30 08:04:37,753 - Recv: T:174.0 E:0 W:?
2017-05-30 08:04:38,755 - Recv: T:174.5 E:0 W:?
2017-05-30 08:04:39,757 - Recv: T:175.3 E:0 W:?
2017-05-30 08:04:40,759 - Recv: T:175.9 E:0 W:?
2017-05-30 08:04:41,761 - Recv: T:176.5 E:0 W:?
2017-05-30 08:04:42,762 - Recv: T:177.2 E:0 W:?
2017-05-30 08:04:43,765 - Recv: T:177.7 E:0 W:?
2017-05-30 08:04:44,767 - Recv: T:178.3 E:0 W:?
2017-05-30 08:04:45,769 - Recv: T:178.9 E:0 W:?
2017-05-30 08:04:46,772 - Recv: T:179.4 E:0 W:?
2017-05-30 08:04:47,773 - Recv: T:180.0 E:0 W:?
2017-05-30 08:04:48,774 - Recv: T:180.7 E:0 W:?
2017-05-30 08:04:49,777 - Recv: T:180.9 E:0 W:?
2017-05-30 08:04:50,778 - Recv: T:181.6 E:0 W:?
2017-05-30 08:04:51,781 - Recv: T:182.0 E:0 W:?
2017-05-30 08:04:52,782 - Recv: T:182.6 E:0 W:?
2017-05-30 08:04:53,785 - Recv: T:183.0 E:0 W:?
2017-05-30 08:04:54,787 - Recv: T:183.5 E:0 W:?
2017-05-30 08:04:55,790 - Recv: T:184.0 E:0 W:?
2017-05-30 08:04:56,793 - Recv: T:184.4 E:0 W:?
2017-05-30 08:04:57,795 - Recv: T:184.8 E:0 W:?
2017-05-30 08:04:58,797 - Recv: T:185.3 E:0 W:?
2017-05-30 08:04:59,803 - Recv: T:186.3 E:0 W:?
2017-05-30 08:05:00,802 - Recv: T:186.9 E:0 W:?
2017-05-30 08:05:01,804 - Recv: T:187.5 E:0 W:?
2017-05-30 08:05:02,807 - Recv: T:188.2 E:0 W:?
2017-05-30 08:05:03,810 - Recv: T:188.7 E:0 W:?
2017-05-30 08:05:04,812 - Recv: T:189.0 E:0 W:?
2017-05-30 08:05:05,815 - Recv: T:190.2 E:0 W:?
2017-05-30 08:05:06,817 - Recv: T:190.5 E:0 W:?
2017-05-30 08:05:07,822 - Recv: T:191.1 E:0 W:?
2017-05-30 08:05:08,821 - Recv: T:191.8 E:0 W:?
2017-05-30 08:05:09,823 - Recv: T:192.1 E:0 W:?
2017-05-30 08:05:10,825 - Recv: T:192.6 E:0 W:?
2017-05-30 08:05:11,827 - Recv: T:193.2 E:0 W:?
2017-05-30 08:05:12,829 - Recv: T:193.8 E:0 W:?
2017-05-30 08:05:13,831 - Recv: T:194.3 E:0 W:?
2017-05-30 08:05:14,833 - Recv: T:194.8 E:0 W:?
2017-05-30 08:05:15,835 - Recv: T:195.2 E:0 W:?
2017-05-30 08:05:16,837 - Recv: T:195.7 E:0 W:?
2017-05-30 08:05:17,839 - Recv: T:196.1 E:0 W:?
2017-05-30 08:05:18,841 - Recv: T:196.4 E:0 W:?
2017-05-30 08:05:19,843 - Recv: T:197.1 E:0 W:?
2017-05-30 08:05:20,845 - Recv: T:197.5 E:0 W:?
2017-05-30 08:05:21,846 - Recv: T:197.9 E:0 W:?
2017-05-30 08:05:22,849 - Recv: T:198.6 E:0 W:?
2017-05-30 08:05:23,851 - Recv: T:198.8 E:0 W:?
2017-05-30 08:05:24,853 - Recv: T:198.9 E:0 W:?
2017-05-30 08:05:25,854 - Recv: T:199.7 E:0 W:?
2017-05-30 08:05:26,857 - Recv: T:200.3 E:0 W:?
2017-05-30 08:05:27,862 - Recv: T:200.4 E:0 W:?
2017-05-30 08:05:28,861 - Recv: T:201.2 E:0 W:?
2017-05-30 08:05:29,862 - Recv: T:201.1 E:0 W:?
2017-05-30 08:05:30,865 - Recv: T:202.1 E:0 W:?
2017-05-30 08:05:31,866 - Recv: T:202.0 E:0 W:?
2017-05-30 08:05:32,869 - Recv: T:202.0 E:0 W:?
2017-05-30 08:05:33,870 - Recv: T:202.3 E:0 W:?
2017-05-30 08:05:34,873 - Recv: T:202.1 E:0 W:?
2017-05-30 08:05:35,875 - Recv: T:202.0 E:0 W:?
2017-05-30 08:05:36,878 - Recv: T:201.7 E:0 W:?
2017-05-30 08:05:37,880 - Recv: T:201.1 E:0 W:?
2017-05-30 08:05:38,883 - Recv: T:200.8 E:0 W:?
2017-05-30 08:05:39,885 - Recv: T:200.4 E:0 W:?
2017-05-30 08:05:40,888 - Recv: T:199.6 E:0 W:?
2017-05-30 08:05:41,890 - Recv: T:199.3 E:0 W:?
2017-05-30 08:05:42,896 - Recv: T:198.9 E:0 W:?
2017-05-30 08:05:43,895 - Recv: T:198.3 E:0 W:?
2017-05-30 08:05:44,898 - Recv: T:198.0 E:0 W:?
2017-05-30 08:05:45,901 - Recv: T:197.5 E:0 W:?
2017-05-30 08:05:46,903 - Recv: T:197.4 E:0 W:?
2017-05-30 08:05:47,905 - Recv: T:197.0 E:0 W:?
2017-05-30 08:05:48,907 - Recv: T:197.0 E:0 W:?
2017-05-30 08:05:49,909 - Recv: T:197.0 E:0 W:?
2017-05-30 08:05:50,911 - Recv: T:197.0 E:0 W:?
2017-05-30 08:05:51,913 - Recv: T:197.0 E:0 W:?
2017-05-30 08:05:52,915 - Recv: T:197.1 E:0 W:?
2017-05-30 08:05:53,917 - Recv: T:197.1 E:0 W:?
2017-05-30 08:05:54,919 - Recv: T:197.5 E:0 W:?
[...]
```

There's nothing that OctoPrint can do about that. All it can do is send the commands you tell it to, here that's a `M109 S205 T0`, and it requires your firmware to properly execute said commands. This doesn't happen here, your firmware first heats, then stops heating for whatever reason, letting the hotend cool down again, then back to heating, etc...

Sorry, but OctoPrint can't do anything here, it's an issue with your printer and/or its firmware.

See also [this entry in the FAQ](https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-wont-cancel-when-my-printer-is-heating-up) which is somewhat related to this.

Qixuga, 2017-05-30:
OK, i thought it might be the printer.
I will contact the developer - thank you!

foosel, 2017-05-30:
Perfect! Closing this then :)

closed 2017-05-30 by foosel