OctoPrint/OctoPrint #1940: [Request] Option to not capture post roll if print is cancelled

gruvin, 2017-05-30:
I have read the FAQ. :-)

Every time I cancel a print, Octoprint starts a lengthy time lapse post roll capture process, reporting times to complete exceeding an hour. These add up pretty quick at times, loading down the system.

An option in the Timelapse tab to not process post roll (and delete the images) if a print is cancelled would seem a very good idea to me. ;-)

Thanks!

Patreon Supporter <-- because, you know .. that's just make me super special and stuff. :-P

fiveangle, 2017-06-10:
This feature would actually be a useful enhancement, since having to deal with deleting all of the failed print TLs is an uncessary annoyance.

Good suggestion !

-=dave

ntoff, 2017-06-13:
+1

I've heard arguments that "but the post roll can provide vital evidence of what went wrong", to that I say if you're looking at octotprint's interface long enough to see something is wrong and hit cancel, whatever caused it has already been captured on "film".

I'd suggest not deleting the timelapses though, the new multi select option for timelapses makes deleting multiple timelapses quite easy, and having the timelapse can help determine what went wrong *before* the cancel button was pressed.

gruvin, 2017-06-13:
That's a good point. Didn't think of that. Maybe all we need is the option in the Timelapse system to cancel currently running jobs. People who don't know or care will get what they have now. More 'sysadmin' types will be able to go in there and cancel those reallllly long background tasks, via the UI.

devsfan1830, 2017-06-17:
+1 id absolutely love this option. Maybe make it a toggle in the settings to enable or disable a cancel timelapse/postroll feature for those who want to keep it as it is. Let us opt in if we dont care about playing print fail CSI via timelapses.

edqallen, 2020-06-28:
+1 to the option.  Just something in the timelapse settings that you can enable to not create timelapses if a print is canceled would be great.

alandsidel, 2020-06-28:
+1  I'd love to have an option in the timelapse settings to not create them if the print is canceled, and to just discard the intermediate data.

gruvin, 2021-09-09:
Seems the only way to get this ancient thing of my open issues list is to close it. Weird. 🤷

closed 2021-09-09 by gruvin

foosel, 2021-09-09:
Well... A PR would also have been an option.

fiveangle, 2021-09-09:
> Seems the only way to get this ancient thing of[f] my open issues list is to close it.

The OCD struggle is real 😉