OctoPrint/OctoPrint #1941: Time out after G4

gmccauley, 2017-05-31:
#### What were you doing?

The MP Select Mini will turn the fan on while the hotend is above a certain temperature without turning it back off it seems.  I've heard of folks putting a G4 (dwell) command at the end of their prints to allow the hotend to cool and then turning the fan off.  Just like Issue #1506, the MP Select Mini is returning an ok immediately and then dwelling causing communication timeouts, then disconnects.

#### What did you expect to happen and what happened instead?

What I expected:  OctoPrint to pause and wait the 4 minutes before attempting to turn off the fan.
What happened:  Printer returned an ok after G4 and OctoPrint tried to turn the fan off while the printer was dwelling

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.2 (master branch

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

MP Select Mini
Send: N1 M115*39
Recv: NAME: Malyan\x09VER: 2.9\x09MODEL: M200\x09HW: HA02

#### Browser and Version of Browser, Operating System running Browser

Google Chrome
Version 58.0.3029.110 (64-bit)

#### Link to contents of terminal tab or serial.log

Send: N53045 M84*24
Recv: ok N53044 P14 B15 T:179.0 /0.0 B:59.6 /0.0 T0:179.0 /0.0 @:0 B@:0
Send: N53046 G4 P240000*95
Recv: ok N53046 P15 B14
Send: N53047 M106 S0*98
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N53048 M105*45
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N53049 M105*44
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N53050 M105*36
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N53051 M105*37
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N53052 M105*38
No response from printer after 6 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from 'Printing' to 'Offline: Too many consecutive timeouts, printer still connected and alive?'
Connection closed, closing down monitor

I have read the FAQ.

foosel, 2017-05-31:
Would be great, because if that setting solves at least the "oh noes, timeout!" from showing up (it really should), I'll add that to the detection for 1.3.4.

gmccauley, 2017-05-31:
Well, it kinda worked?  It waited for 4 minutes after the G4 before it displayed the ok and then executed M106 and received an ok and then changed the state to operational.  It never displays that it tried to execute any M105's to get temperature though and then after about 2 more minutes, it says that there were communication timeouts.

When I checked on the printer, the fan was still running, but since it never issued an M105 after the dwell, I'm not even sure that the hotend was below the required temperature, so it's possible that OctoPrint turned the fan off and then the firmware turned it back on.

Terminal Output:
Send: N6233 M84*43
Recv: ok N6233 P15 B15
Send: N6234 G4 P240000*104
Recv: ok N6234 P15 B15
Send: N6235 M106 S0*85
Recv: ok N6235 P15 B15
Changing monitoring state from 'Printing' to 'Operational'
No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from 'Operational' to 'Offline: Too many consecutive timeouts, printer still connected and alive?'
Connection closed, closing down monitor

gmccauley, 2017-06-01:
Just for S&G, I rebooted the Pi after making those changes, and now it is working as expected.

Terminal Output:
Send: N6232 M84*42
Recv: ok N6232 P15 B15
Send: N6233 G4 P240000*111
Recv: ok N6233 P15 B15
Send: N6234 M106 S0*84
Recv: ok N6234 P15 B15
Changing monitoring state from 'Printing' to 'Operational'
Send: N6235 M105*21
Recv: ok N6235 P15 B15 T:72.5 /0.0 B:41.5 /0.0 T0:72.5 /0.0 @:0 B@:0

foosel, 2017-06-02:
Oops, forgot to comment, only mentioned in the above commit: The "block while dwelling" feature flag is now also automatically applied when a Malyan printer is detected. 

Already on `maintenance` and `devel`, will be released with 1.3.5.

foosel, 2017-10-17:
1.3.5 was released yesterday.

closed 2017-10-17 by foosel