OctoPrint/OctoPrint #1942: Error starting the new 1.3.3 (master branch) version

FabioMC, 2017-05-31:
I just update to the new 1.3.3 (master branch) but just after it complete the process I was not able anymore to start Octoprint. I send my log file. If someone could help I would really appreciate since I am not an expert.

Thanks

[octoprint.txt](https://github.com/foosel/OctoPrint/files/1043159/octoprint.txt)

johnrosenbaum, 2017-06-01:
Had the same issue. I deleted the Serial section of my config.yaml and it seems to have fixed the issue. Not sure why this is happening though.

foosel, 2017-06-01:
This looks like something is really wrong with the default printer profile at the time of auto-connect on server startup, causing things to explode. However I have no idea HOW this happens in the first place, since I cannot for the life of me reproduce this scenario/get the printer profile to break such that this error can even trigger.

@FabioMC @johnrosenbaum Could you please provide me with your `config.yaml` or at least its `printerProfile` subsection, and `.octoprint/printerProfiles/_default.profile` files? Just send them to gina@octoprint.org please. No need to post them publicly here since at least `config.yaml` might also contain sensitive data.

brentmc79, 2017-06-01:
Had the same issue also, but in my case there appears to be no default profile, or any profiles for that matter. The _default.profile file is blank and the profiles section of the config.yaml is just an empty hash "{}". I was able to get octoprint to start up by physically disconnecting the usb cable, but I'm unable to add new profiles via the UI.

foosel, 2017-06-01:
Ah, that "empty hash" issue might be the clue to at least allow me reproduction. Thank you for that.

In the meantime I've set the release to "pre release" status, that should at least stop the rollout until a hotfix is ready.

foosel, 2017-06-01:
Ok, I now at least have a reproduction. What I still don't understand is why `_default.profile` becomes an empty file in the first place, because it should either be the hard coded default profile or whatever you had previously in `config.yaml`. I've tried to get it to write an empty file by manipulating the `config.yaml` value and running the migration but so far all I get are perfectly valid files.

Any details on how you had your printer profile setup initially (if at all) might help here.

foosel, 2017-06-01:
@brentmc79 could you send me your empty `_default.profile`? It might be important to know whether it is indeed completely empty or if there is some whitespace in there.

brentmc79, 2017-06-01:
@foosel just sent you an email with the file zipped up since gmail wouldn't let me attach a 0 byte file

foosel, 2017-06-01:
Thanks, got it. So it really is completely empty. Huh. I have *no* idea how that could happen. Will try some more to reproduce this particular scenario, but if I don't find that until noon I guess I'll just go with pushing a patch (either as 1.3.3post1 or 1.3.4, not sure yet) that at least solves the crashing issue and fixes the `_default.profile` if it becomes invalid like this.

foosel, 2017-06-01:
@brentmc79 quick questions, what creation and modification dates does the empty profile have on the file system? Not sure if the date it shows me in the zipped version is accurate or not.

brentmc79, 2017-06-01:
```
pi@octocake:~/.octoprint/printerProfiles $ stat _default.profile 
  File: ‘_default.profile’
  Size: 0         	Blocks: 0          IO Block: 4096   regular empty file
Device: b302h/45826d	Inode: 44265       Links: 1
Access: (0644/-rw-r--r--)  Uid: ( 1000/      pi)   Gid: ( 1000/      pi)
Access: 2017-06-01 03:14:39.634282687 +0000
Modify: 2017-06-01 03:14:39.644282618 +0000
Change: 2017-06-01 03:14:39.644282618 +0000
 Birth: -
```

foosel, 2017-06-01:
Thank you. That pretty much rules out the idea I had while stepping through various versions of past code that it might have been caused by an old `_default.profile` still lying around. So that idea was a dead end, which is both a relief and a curse ;)

ctgreybeard, 2017-06-01:
is `printerProfiles: {}` incorrect? That is how my config.yaml reads. I don't use auto connect and I haven't experienced any problems. I do have one profile named Prusa configured. _default.profile exists and appears to be fully formed.

foosel, 2017-06-01:
`printerProfiles: {}` is fine. But something like

```
printerProfiles:
  defaultProfile: null
```

wouldn't be.

I just released 1.3.4 with a fix for the crashing issue caused by the corrupted profile. I still have not found any way to reproduce it becoming blank. Now I'm a bit unsure what to do about this ticket, since the core reason of the problems is not fixed, only the symptoms are made less severe.

foosel, 2017-06-01:
Ah, before I forget:

### What to do if you are affected by this and your OctoPrint server isn't starting up anymore

  * Unplug your printer from your OctoPrint server (e.g. your Pi)
  * Restart your server, it should now actually start through
  * Update to 1.3.4 (if you do not get the notification that the update is available, go to Settings > Software Update > Advanced Options, click the button that says "Force check for update").
  * Plug your printer back in, all should be working fine now again.

JohnOCFII, 2017-06-01:
Just a related note - 1.3.3 started up for me, but when I tried to print, I received the warning that my print was outside the bounds of the print area.  I noticed that my previously selected printer was no longer selected.  Once I re-selected my printer, all was OK with 1.3.3.

foosel, 2017-06-01:
Yeah, that's tracked in #1943. Sadly did notice that bug only just right after I'd pushed out 1.3.4 (as in, less than 5 minutes after) so now that will have to wait until 1.3.5. Bloody typos ;)

All I can say is, we apparently still need more people testing the release candidates ^^

JohnOCFII, 2017-06-01:
> All I can say is, we apparently still need more people testing the release candidates ^^

Sorry about that.  I had loaded the RCs for 1.3.3 to my Raspberry Pi , but never got as far as connecting to my printer before the released version came out :(

foosel, 2017-06-01:
Wasn't meant as a specific complaint, more a Gerald observation :)

Btw, I usually leave one week between an RC and the next version and always try to leave a full weekend in the middle, just as a rough house number of when to expect releases/new RCs.

foosel, 2019-11-05:
> I just released 1.3.4 with a fix for the crashing issue caused by the corrupted profile. I still have not found any way to reproduce it becoming blank. Now I'm a bit unsure what to do about this ticket, since the core reason of the problems is not fixed, only the symptoms are made less severe.

Decided to close it since I also haven't heard anything about it anymore.

closed 2019-11-05 by foosel