OctoPrint/OctoPrint #1944: [Request] GCode Script: After Heated

Givemeyourgits, 2017-06-01:
#### What were you doing?

Want to set a prime line when jobs start.

#### What did you expect to happen?

I used "Before print job starts" but this is prior to hot end/bed heating up, which means the prime line is done cold.

#### What happened instead?

Hot End/Bed heat up, then run script to print prime line

#### Did the same happen when running OctoPrint in safe mode?

N/A

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.3 (master branch)

#### Operating System running OctoPrint

OctoPi 0.13.0

#### Printer model & used firmware incl. version

N/A

#### Browser and Version of Browser, Operating System running Browser

N/A

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

I have read the FAQ.

FHeilmann, 2017-07-04:
Why not do it in the start script of your Slicer?

atar-axis, 2019-02-26:
I would love that too! Doing it in the slicer is an option but I would prefer centralized behaviour control of the printer.

hayden-t, 2022-06-30:
yes im trying to work out a way to do this, as have a nozzle wipe script once hot right before print start, i can add it the cura, but would prefer everything in one place, maybe there is a way for octoprint to append to start of uploaded gcode that would give same effect as uploaded gcode does not start until hot and printing

hayden-t, 2022-07-02:
yes likely, i also found https://plugins.octoprint.org/plugins/gcode_macro/ which is working well enough, i just put the @macroname into my cura start section and that triggers my script stored in octoprint