OctoPrint/OctoPrint #1945: [Request] expose ui option to disable "reset on connect"

Andrei-Pozolotin, 2017-06-02:
please expose ui option to disable "reset on connect"

https://github.com/foosel/OctoPrint/wiki/FAQ#my-sd-prints-get-interrupted-when-i-connect-to-my-printer-from-octoprint

----------------
Nonetheless, the following works for some boards:
```
stty -F /dev/ttyACM0 -hupcl
```
which will disable "the usual" USB -> Serial -> CTS/DTR, etc -> RESET action

foosel, 2017-06-14:
As far as I understand, that is currently not possible in a cross platform kind of way due to implementation issues in PySerial. And a platform specific way like utilizing `stty` is out of the question in the core application and would require a plugin.

aequitas, 2017-07-23:
Would you consider merging a change to core to add a configurable command/script to run at server startup like `Restart OctoPrint`, `Restart system` and `Shutdown system` under the Octoprint->Server preference? 

This command could then be configured in this place. It would keep the core cross platform but allow to use this fix without writing a separate plugin for it or having to fix it outside of Octoprint.

foosel, 2017-07-24:
But how would having to first configure such a platform specific command differ in "things the user needs to do" from installing a plugin via the plugin manager once?

Plugins can run stuff on server startup already. And it would even already be possible to have a system command excute via a `system` event handler for the [Startup event](http://docs.octoprint.org/en/master/events/index.html#server), e.g. something like

``` yaml
events:
  subscriptions:
  - event: Startup
    command: stty -F /dev/ttyACM0 -hupcl
    type: system
```

in `config.yaml` would probably already work.

aequitas, 2017-07-24:
I really need to read up on the docs for configuration as i'm missing out on all these great features :)

I tried your suggested solution and it works as expected. And it keept this nicely within the Octoprint realm which makes for easier management and backup.

As for not putting it in a plugin I have a tendency to install as little plugins as possible and I felt this feature was really to small to justify a plugin. But I'm not completely familiar with the Octoprint ecosystem yet so I have something to learn there as well.

kami8607, 2017-12-12:
My solution is to add the command as a udev rule which gets executed every time the printer is connected. You can find the values for serial, idVendor and idProduct with lsusb.

/etc/udev/rules.d/96-stty-hupcl.rules:
SUBSYSTEM=="tty", ATTRS{serial}=="A50285BI", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ACTION=="add", RUN+="/bin/stty -F %E{DEVNAME} -hupcl"

cp2004, 2021-05-18:
This one appears to be solved to me, using custom controls/system commands menus.

closed 2021-05-18 by cp2004