OctoPrint/OctoPrint #1948: [BUGs] Zoom not working // Model too big to fit???

bolsoncerrado, 2017-06-02:
----

#### What were you doing?

[Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.

Ideally provide exact steps to follow in order to reproduce your problem:

1. ... Upload model on the Pi
2. ... Start print
3. ... Issue arises

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?

Start printing

#### What happened instead?

It does start printing but I can't either Zoom the model on the GCode viewer and a Warning pops in "Model too big to print" when its clearly not that big!

#### Branch & Commit or Version of OctoPrint

1.3.4 but also had it on 1.3.2

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Operating System running OctoPrint

Raspbian Jessie Octopi

[OctoPi, Linux, Windows, MacOS, something else? With version please.
OctoPi's version can be found in /etc/octopi_version]

#### Printer model & used firmware incl. version

Prusa MK2S
[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser

Chrome on MacOS
[If applicable, always include if unsure.]

#### Screenshot(s)/video(s) showing the problem:

http://i.imgur.com/CwlCWwG.png
http://i.imgur.com/ya042fD.png
http://i.imgur.com/uJnbjNB.png
[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

foosel, 2017-06-02:
Please test if the problem persists if you delete the file in question and reupload it (the file analysis data is most likely still from 1.3.2 which still had a bug if `G92` offsets were used, and file analysis data does NOT get deleted on update). SO do this:

1. delete the file
2. re-upload it, let file analysis finish
3. select it for printing, check if the warning persists

Also make sure your proper printer profile is selected, because it might not be thanks to #1943:

1. Disconnect
2. Verify your printer profile is selected
3. Reconnect

If that doesn't help:

1. provide a file with which to reproduce
2. provide you *currently selected* printer profile

and in the future also provide logs. They are part of the ticket template for a reason and not providing them makes it less likely that your issue will be analysed in a timely manner since I frankly I'm a bit tired of having to run after information all the time.

bolsoncerrado, 2017-06-02:
Seems reuploading did the trick. Never faced this in the 3+yr ive been using the tool hence I was quite surprised.

I still get mixed results on the zoom issue though, I think it's related on me clicking the checkboxes BEFORE the analyis ends... it's really hard to find a pattern to reproduce it, but I tend (if it helps) to uncheck travel moves / retracts, enable next layer, center, and zoom while the model is being analyzed. Perhaps the zoom in routine is "loaded" next, collapsing the experience?

louvanna, 2017-06-03:
I also had a too big for printer, outcome. This is  seen for the first time on 1.34. A model which was previously running fine on 1.32.
Delete,& Re-load, did indeed not repeat this message.
But there is an artifact which is persistent. The gCode viewer puts the model in the lower left corner of the grid, where center of model is at the exact corner of the grid.

Printing it anyway, results in a centered item on table.

Octoprint SD image, recently updated to 1.34,
RostockMax v3 printer

bolsoncerrado, 2017-06-03:
Yes I noticed the uncentered zoom too...

foosel, 2017-06-06:
To quote myself from up there again:

> Also make sure your proper printer profile is selected, because it might not be thanks to #1943:
>
>   * Disconnect
>   * **Verify your printer profile is selected**
>   * Reconnect

To clarify, #1943 can cause your default printer profile to not be selected when you connect. And your described symptoms (apart from the zoom issue I've not been able to reproduce myself so far) sound like exactly that being the case here.

louvanna, 2017-06-09:
After setting the profile to 'my' profile, #1943, the gCode-layer viewer returns to normal.

Thanks.

louis.

bolsoncerrado, 2017-06-09:
just commenting ive had this issue twice on different models (even after reuploading them) on 1.3.4

foosel, 2017-06-09:
@bolsoncerrado and yet you make absolutely sure that I have no chance to look into this by refusing to provide any more information than "It's still happening!". No logs, no GCODE files and profile information to attempt reproduction, not even feedback whether you tried any of the steps I outlined above and checked that you indeed are on the correct printer profile.

At this point it really feels like you are doing this on purpose ☹️

bolsoncerrado, 2017-06-09:
Oh no i dont! I just thought I understood you sorted this out already on 1.3.5?? My bad if you didnt!

foosel, 2017-06-09:
@bolsoncerrado I fixed #1943 in 1.3.5. You said you saw "this issue" (which one exactly, zoom or "too big"?) on 1.3.4, twice. So the way I'm seeing it, you tried what I suggested earlier and saw that it did NOT fix your problem, so you are apparently seeing a problem (or a cause for it) that I'm not aware of and expect me to do something about it because otherwise no point in commenting, right? But for actually looking into it you don't provide me with any info besides "it's still broken". See how this could be frustrating?

Please provide the following for your "too large" issue:

  * `octoprint.log` when the problem arises
  * your printer profile(s) with which you are seeing your "too large" issue
  * your gcode files(s) with which you are seeing your "too large" issue

And for the zoom problem:

  * Javascript console when the problem arises
  * a video showing the problem (because I still don't know exactly what I'm even looking for)

I don't sit in front of your screen. I can't see what you see. I have to depend on what you describe to me. So when I tell you "give me more info", I'm not doing that to keep you busy but because your description so far was not sufficient for me to get a clear picture or an inkling of an idea of what might be happening.

bolsoncerrado, 2017-06-09:
Im still on 1.3.4 thats why I posted in case someone else had this issue... And yeah, talking about the TOO BIG one, sorry. I definetly need a rest.

bolsoncerrado, 2017-06-09:
So for the record, yes, printer profile is correctly assigned and machine correctly defined.

I can't record a video of the TOO BIG because it's quite of random, but next time I'm able to reproduce it more than once, will try to.

As for the Zoom issue, not sure if this can help at all?

https://youtu.be/8FQ9ip9l2Tw

foosel, 2017-06-09:
About the zoom issue, at least now I have an idea of what you are seeing which so far I didn't, so that's definitely a step forward :)

foosel, 2017-06-09:
Can you also provide a file you are seeing this with?

bolsoncerrado, 2017-06-09:
Perhaps try with these?

https://www.dropbox.com/s/nnz59my4mrj2jtp/C_608_BEARING_MOUNT_v2_098pct.gcode
https://www.dropbox.com/s/swjs4qmkual05sr/kkkkkC_608_BEARING_MOUNT_v2_0.gcode

foosel, 2017-06-13:
The only zoom issue I could reproduce with those is simply caused by your nozzle priming routine happening outside of your print bed. OctoPrint has no way to distinguish between extrusion for nozzle priming and extrusion that is actually part of building up your printed part. So it considers any kind of extrusion + movement happening as part of the model and hence the coordinates it is happening at as model coordinates, meaning if you have a priming procedure happening outside of the bed, the bounding box of your model (which is what's used to center on your model and also to zoom in on your model) will include that priming position and that will lead to wonky center coordinates.

To maybe explain this a bit better, a screenshot with the bounding box for any kind of extrusion + movement happening in blue and the bounding box excluding any such stuff that happens outside of the bed (in the priming zone at y = -3) in red:

[image]

I'm currently experimenting with just ignoring any kind of extrusion + movement outside of the bed bounds and am thinking about allowing to define a "priming zone" in the printer profile to avoid visualization issues like this.

But in general zooming and panning around is working just fine for me with these two files (while idle and while printing), so no reproduction of it not working at all so far. Also still no reproduction of the "too large" issue either (with confirmed correctly configured printer profile to circumvent #1943).

foosel, 2019-11-05:
Closing this since it's still unreproduced (for correctly configured printer profiles).

closed 2019-11-05 by foosel