OctoPrint/OctoPrint #1950: Octoprint completely broken and not responding

RodneyMcKay715, 2017-06-02:
#### What were you doing?

I used OctoPrint normally for a month with the 1.3.2 release and everyting was woring fine up to this point.

#### What did you expect to happen?

To be able to print.

#### What happened instead?

- Octoprint is insanely slow reacting (we are talking in multiple minutes for a tab change)
- Timelapse Tab has no content
- Cancel button has no function
- Webcam is accessible via link, but not under Webcam tab
- Accessing the Webcam tab causes the Pi to crash entirely
- Prints get distorted and skip alot of layers.
- Z- Offset gets ignored
- Files cant be uploaded properly, manual deletion via putty needed.
- If a Timelapse finishes rendering the Server shuts down and won't come back up
- Connecting to a Printer takes 5-10 Minutes
- After first start of Octoprint you need to restart Octoprint because the Upload Overlay is stuck and wont go away, even after a F5 is done
- Integrated Slicer is producing empty models
- Terminal Tab is not accessible and corrupts the config.yaml when selected (not always)

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Branch & Commit or Version of OctoPrint

1.3.4

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

Anet A6 Skynet 2.3.2

#### Browser and Version of Browser, Operating System running Browser

Firefox, Chrome, Internet Explorer, Edge, Opera, Safai 8all the newest ones)

#### Link to octoprint.log

https://gist.github.com/RodneyMcKay715/1ddc5fc29a4823db34a3e31442daef75

#### Link to contents of terminal tab or serial.log

https://gist.github.com/RodneyMcKay715/44121be5c0a55d361164c1401f4cbe27

#### Link to contents of Javascript console in the browser

https://gist.github.com/RodneyMcKay715/e448fdadf42283d4430ff72a14c97243

markwal, 2017-06-02:
Your logs point at several issues on your end that you'll want to look into.

First, it appears that you've set your timelapse folder (and thus the temporary image folder as well) to a removable media (usb stick?).  There are several problems with doing this, but one is that OctoPrint doesn't expect that folder to change out from underneath it while it is running.  It's not clear if you also have your upload folder pointed at that too, but the having the upload folder change out from underneath OctoPrint can cause some problems like having to reanalyze gcode files which can't take a significant amount of processing time.  If the USB stick is just for extra storage and you aren't intending to remove it and fiddle with the filesystem, that's fine, but then there are indications of other problems here:

There's something wrong with the file system on that USB stick.  Some of the errors are indicative of corruption, or it being set to read-only, or out of space.  Another possibility is that there is corruption in your config.yaml.  You may want to check that file.  It's normally in /home/pi/.octoprint/config.yaml.

In your shoes, I'd make sure the microSD card has sufficient space, redirect the folders back to the default locations and see if that resolves the problem.  Failing that you may want to start with a clean config.yaml.  If you'd like me to take a look at the config.yaml, you can post it like the others (on gist), but make sure you redact any secrets from it first (accessControl salt, api key, server secretKey and any passwords for services that plugins might be connected to [telegraph, dropbox, youtube, etc.])

foosel, 2017-06-23:
Any update on this?

RodneyMcKay715, 2017-06-23:
I'm currently printing over SD card, since Octopi won't even work after a complete fresh install. I install it, it runs and then if i select a menu the Octoprint Server is shut down and i can't restart it. Maybe my Pi is broken? Idk. I'm waiting for the next OctoPi Release and try again then.

markwal, 2017-06-23:
@RodneyMcKay715 On your fresh install where the server shutdown and won't restart, could you grab the log from that and post it like you did the others?  I assume the log is still in that state?

If there is a hardware problem, it's probably the power supply, not the pi.

Though, that's because I'm assuming you are verifying that the server isn't running and won't restart via the command line.  If what you mean instead is that the web UI has the pop-up box that says it is disconnected whenever you try to do anything, it is more likely that the server is running fine, but wifi is browning out either due to emf noise nearby or again power problems.  If so, a new version of Octoprint won't change anything.

RodneyMcKay715, 2017-07-27:
A little Update on this. I got it fixed. I had the Pi turned off for about 3 weeks after reinstalling over the image and configuring it. I turned it on today to try to fix it, but no problems were present. I will have an eye on my pi but for now everything works as it should. Definitely not a OctoPrint issue. Either my Powerstrip is acting up (had this with my PC) or the Pi's Power Supply. But currently everything works fine.
How can i check for EMF Noise btw?

markwal, 2017-07-27:
Most common symptom of emf noise problems is lost bytes or extra garbage over the serial wire. You can see this in the terminal window. Lots of resends, etc.  And timeouts and disconnects.

closed 2017-07-27 by markwal