OctoPrint/OctoPrint #1952: [Request] Anet A8 hotend disabled

presidioh24, 2017-06-03:
#### What were you doing?

1. Export Slic3r configuration ini file for ABS & Anet A8 mk2
2. Import into Octopi Slic3r settings (I need to retry many times with reboot to do it)
3. Slicing a valid .stl project with Slic3r (Already sliced and printed with CuraEngine successfully)
4. Load & Print

#### What did you expect to happen?

Print the object

#### What happened instead?

Before starting, during the preheating pahse, the HOTEND go to the configured temperatures for 1 seconds and then go to OFF.
The hotbed preheating correctly

When i click on CANCEL button the hotbed still be hot and no cooldown process start.
This issue doesn't occour when I do cancel with a gcode sliced by CuraEngine.

This occours with all tested profile ( 4 profiles)

#### Did the same happen when running OctoPrint in safe mode?

Can't use slic3r in safe mode (if I'm not in wrong)

#### Branch & Commit or Version of OctoPrint

latest Version 1.3.4 (master branch)

#### Operating System running OctoPrint

Linux octopi 4.9.24-v7+ #993 SMP Wed Apr 26 18:01:23 BST 2017 armv7l GNU/Linux
pip 9.0.1 from /home/pi/oprint/local/lib/python2.7/site-packages (python 2.7)
Rpi 3

#### Printer model & used firmware incl. version

Anet A8 mk2 i3 - original firmware

#### Browser and Version of Browser, Operating System running Browser

CHROME Versione 58.0.3029.110 (64-bit)
Windows 7 pro 64bit

#### Link to octoprint.log

https://gist.github.com/presidioh24/d005a433d16a4b330b2488fd685acfce

#### Link to contents of terminal tab or serial.log

https://gist.github.com/presidioh24/41aecfea90d8d2c53aea7873fea35264

#### Link to contents of Javascript console in the browser

unable to do that
#### Screenshot(s)/video(s) showing the problem:

SLIC3R PROFILE
https://gist.github.com/presidioh24/5b8ede515d7d19d35fb462fb94a1b4c4
https://gist.github.com/presidioh24/2e94572b65596152e21c78e7585d3adb
https://gist.github.com/presidioh24/5f1ad9f3ece047e29681a88c88994c70

Screenshot about the problem
http://it.tinypic.com/r/5we89v/9

ntoff, 2017-06-04:
Might want to remove your profiles, they contain your octoprint API key. Better delete those and re-upload.

apart from that
>G1 Z+0.5 E-5 X-20 Y-20 F{travel_speed} 

F{travel_speed} is a cura variable, not sure that'll do you much good in slic3r (I know the regular version of slic3r uses square braces for variables, assume that's also the case for the plugin). Check your start and end scripts don't contain cura specific variables.

presidioh24, 2017-06-05:
Hi, 
thank you for the fast replay.

*After your reply, I've opened Slic3r from my pc and then exported this:

avoid_crossing_perimeters = 1
bed_size = 200,200
bed_temperature = 96
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 60
brim_width = 0
complete_objects = 0
cooling = 1
default_acceleration = 0
disable_fan_first_layers = 1
dont_support_bridges = 1
duplicate_distance = 6
end_gcode = M104 S0 ; turn off temperature\nG28 X0  ; home X axis\nM84     ; disable motors\n
external_perimeter_speed = 70%
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0
fan_always_on = 0
fan_below_layer_time = 60
filament_diameter = 1.75
fill_angle = 45
fill_density = 80%
fill_pattern = honeycomb
first_layer_acceleration = 0
first_layer_bed_temperature = 101
first_layer_extrusion_width = 200%
first_layer_height = 0.4
first_layer_speed = 30%
first_layer_temperature = 255
g0 = 0
gap_fill_speed = 20
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 0
infill_every_layers = 10
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
infill_speed = 60
interface_shells = 0
layer_gcode = 
layer_height = 0.3
max_fan_speed = 100
min_fan_speed = 35
min_print_speed = 10
min_skirt_length = 0
notes = 
nozzle_diameter = 0.4
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 30
perimeters = 3
post_process = 
print_center = 100,100
raft_layers = 0
resolution = 0
retract_before_travel = 2
retract_layer_change = 1
retract_length = 1
retract_length_toolchange = 10
retract_lift = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 30
seam_position = aligned
skirt_distance = 6
skirt_height = 1
skirts = 1
slowdown_below_layer_time = 30
small_perimeter_speed = 30
solid_fill_pattern = rectilinear
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extrusion_width = 0
solid_infill_speed = 60
spiral_vase = 0
standby_temperature_delta = -5
start_gcode = G28 ; home all axes\nG1 Z5 F5000 ; lift nozzle\n
support_material = 0
support_material_angle = 0
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 0
support_material_interface_speed = 100%
support_material_pattern = pillars
support_material_spacing = 2.5
support_material_speed = 60
support_material_threshold = 0
temperature = 250
thin_walls = 1
threads = 2
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 50
top_solid_layers = 3
travel_speed = 130
use_firmware_retraction = 0
use_relative_e_distances = 0
vibration_limit = 0
wipe = 0
z_offset = 0

Same issue occours, i'm not able to preheat the hotend and then print

foosel, 2017-06-06:
I'm not sure why you opened a bug/feature request here. The Slic3r plugin is not part of OctoPrint's standard distribution and qualifies as a "third party plugin". So if you have any particular issues with that you should report those [in the repository of the plugin](), not here.

Your whole problem though sounds more like an issue with your slicing profiles or with slic3r itself than with OctoPrint or the OctoPrint Slic3r plugin. The mailinglist or G+ community might be a better location since those are the official places for asking support type of questions.

OctoPrint does not manipulate the GCODE produced by your slicer in any way (well, unless you install plugins that explicitly change that). It won't remove or swap out temperature commands or anything like that. So if your start code in your slicer does not properly heat up your hotend, it's your start code being wrong. I'm also missing any heatup commands at all in your provided profiles, so maybe that's the reason. I don't know how slic3r handles its start code, if it automatically inserts heatup commands and in what way if none are present. If you'd provided an example file (as in, a file that you sliced as explained above) it would also probably be fairly easy to see what is going on here, but sadly you didn't.

Since this is not a bug in OctoPrint or anything at all really that I can do anything about (as I said, OctoPrint sends what's in your GCODE files, you need to make sure what's in there makes sense at another location in your workflow), I'm closing this. Comments are still open though.

closed 2017-06-06 by foosel