OctoPrint/OctoPrint #1953: Problem to start the server

varazir, 2017-06-03:
I have this logs and setup 

https://gist.github.com/varazir/9804ad90ac59e412ea799d2fd9f6a5ea

I can't see any error with it and nothing in the log that points me to a error. 

//Daniel

foosel, 2017-06-06:
Your `/home/pilot/.octoprint/uploads` is a file. It should be a folder. Did you create the folders (or - in your case - files) under `/home/pilot/.octoprint` manually by any chance? You don't need to (and if so please make sure they are *folders*, not *files*, with write permissions for the user you are running OctoPrint under), OctoPrint will do that automatically on first startup if necessary.

If this used to work but now doesn't you are likely facing file system corruption (any cases where you just powered off the machine instead of properly shutting it down first maybe?). 

PS: Please follow the bot's instructions for future tickets.

varazir, 2017-06-06:
Hello, thanks for your replay, should have read the info better. Sorry about that. 

The problem is that I don't want uploads to be on the local drive. I like to be a network drive. 
Found a solution to that, I mounted the network drive uploads/server. 

Same with timelapse, but it worked better with mounting. 

You can close the ticket.

closed 2017-06-06 by foosel