OctoPrint/OctoPrint #1954: Video Streaming Localhost Issue

zenvent, 2017-06-04:
#### What were you doing?
I am setting up the webcam stream for octoprint, as I have done in the past before.
When I configure the webcam, I'm finding it works fine if I use the stream address with a full IP address.
However if I change it to local host it no longer works.
What is truly odd is I can use localhost in the snapshot url, and it works just fine!

My issue is my IP address are dynamic, so they could change at any time. I would like both to use localhost as they used to do.

Examples:
http://localhost:8080/?action=snapshot works in the 'snapshot url test'
http://192.168.1.249:8080/?action=stream works in the stream url (it's the device's ip address)
http://localhost:8080/?action=stream does not work in the stream url. I get a broken box icon.

#### What did you expect to happen?
I expect both snapshot and stream url to be able to use local host, as I have configured mjpg streamer to use localhost 8080.

#### What happened instead?
Nothing, I see a broken box icon, unless I use the devices own full ip address.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Branch & Commit or Version of OctoPrint
Version 1.3.2 (master branch)

#### Operating System running OctoPrint
Armbian, latest.

#### Printer model & used firmware incl. version
N/A

#### Browser and Version of Browser, Operating System running Browser
Chrome & IE.

#### Link to octoprint.log
https://pastebin.com/iHnQYTaf

#### Link to contents of terminal tab or serial.log
There is nothing in my terminal or serial log.

#### Link to contents of Javascript console in the browser
Failed to load resource: net::ERR_CONNECTION_REFUSED
localhost:8080/?action=snapshot&1496556012094 Failed to load resource: net::ERR_CONNECTION_REFUSED
localhost:8080/?action=stream Failed to load resource: net::ERR_CONNECTION_REFUSED
localhost:8080/?action=stream&1496556045743 Failed to load resource: net::ERR_CONNECTION_REFUSED
localhost/ Failed to load resource: net::ERR_CONNECTION_REFUSED

#### Screenshot(s)/video(s) showing the problem:
I have read the FAQ.

nophead, 2017-06-04:
The stream URL should be: /webcam/?action=stream

Snapshots can use localhost because they are grabbed by the OctoPrint server which is running on the local host. It can't be used for the stream because that is used by the client webpage. localhost would the clients IP.

zenvent, 2017-06-04:
I tried '/webcam/?action=stream' and see the same issue. I also tried '/?action=stream'.
This is what my execstart looks like in the mjpg service:
./mjpg_streamer -i "./input_uvc.so" -o "./output_http.so"

nophead, 2017-06-04:
I think you need haproxy setup correctly to use /webcam

zenvent, 2017-06-04:
In my nginx configuration I have set a proxy for the webcam:

location /$SHORTNAME/webcam/ {
            proxy_pass http://127.0.0.1:8080/;
        }
My shortname is just octoprint.

I tried '/octoprint/webcam/?action=stream' and it worked.

You helped lead me to the answer, I was banging my head on the desk for a while there. Funny how computers do that. Thank you.

closed 2017-06-04 by zenvent