OctoPrint/OctoPrint #1955: [Request] Flip webcam for stream URL

shanehull, 2017-06-06:
My camera is installed upside down because of the way it is mounted, and although you can flip the camera horizontally/vertically, this doesn't impact the raw stream URL.

It would be great to be able to control the orientation of this via the UI.

ntoff, 2017-06-06:
The raw stream is only embedded by octoprint and can come from anywhere, it's handled by an external program. It just happens to come from mjpeg streamer running on the same device if you're using octopi (or manually set it up in the same fashion), but can be any external source (even one running on an entirely different computer) and as such octoprint may have very limited to no control over it. 

I believe octoprint gets around this for the timelapse by simply flipping the image using some ffmpeg filters, and the displayed stream by some img html tags, there's nothing it can do in order to control the raw stream sent out via mjpeg streamer.

If you're using a pi camera module you can flip it in the configuration of mjpeg streamer https://github.com/foosel/OctoPrint/wiki/MJPG-Streamer-configuration but that isn't supported for external web cams (at least I couldn't get it to work).

foosel, 2017-06-06:
What @ntoff said. With some fiddling it might be possible to write a plugin for mjpg-streamer configuration specifically tailored towards OctoPi, but it would be a huge hassle. And there's no way to do this in a generic kind of fashion because OctoPrint does only *embed* the webcam, it has absolutely no control at all over it, it can only manipulate its own representation of it.

closed 2017-06-06 by foosel