OctoPrint/OctoPrint #1957: octopi-network.txt incorrect

tskamath, 2017-06-06:
the octopi-network.txt file is incorrect.

## WPA/WPA2 secured
# iface wlan0-octopi inet **manual**
iface wlan0-octopi inet **dhcp**
    wpa-ssid "SSID"
    wpa-psk "password"

Regards

foosel, 2017-06-06:
No, it is not (and this would be the wrong bug tracker if it was).

*Raspbian Jessie* (which OctoPi is based on) does DHCP via a custom daemon that depends on entries it is supposed to handle in `/etc/network/interfaces` and included files to be marked as `manual` instead of the more intuitive `dhcp`. I don't know why they did it that way either, but it's the way things are. If you want to learn more about this, look up dhcpcd on Raspbian.

closed 2017-06-06 by foosel

tskamath, 2017-06-06:
Hi,

I full appreciate the hard work and need to comply, but if the WiFi is to be configured with editing octopi-network.txt  does not connect to network at all.

I am not sure on the details inner working , but the WiFi did not work at all, until the ‘manual' changed it to ‘dhcp’. 

Hence the whole purpose of editing the octopi-network.txt on the SD card makes no sense. Maybe its not a bug, but its also not a configuration edit to make the WiFi work.

Thanks & Regards
Srikanth Kamath,

foosel, 2017-06-06:
It works just fine for thousands of people worldwide who are running the stock OctoPi images (including myself) the way it is, with the `manual` setting, because `dhcpcd` is taking care of things.

tskamath, 2017-06-06:
Dear Gina,

No disputes, but it still does not work for a a minority of people, who download from your website, no changes except editing the octopi-network.txt to enable the WiFi, as detailed in the install guide.

If I was to debug and and find, why the “manual’ to ‘dhcp’ solved the issue, I would have forked but unfortunately my skill set are not adequate, hence I was hoping that maintainers could get to the bottom of this and provide a fix to small minority like me.

Thanks & Regards
Srikanth Kamath,

tskamath, 2017-06-07:
not sure if i have got to bottom of this, the WiFi configuration breaks if 

`sudo ./scripts/install-desktop` else without the desktop installed the default configuration works
`iface wlan0-octopi inet manual`

tskamath, 2017-06-08:
Foundation releases of Raspbian Jessie has changed the configuration  (and particularly WiFi) works, there are lot of confusion in the world "google" / and many ways of doing it and the following url explains it good.. https://raspberrypi.stackexchange.com/questions/37920/how-do-i-set-up-networking-wifi-static-ip-address

the following is documented for anyone else to hit the same issue (one in million like Gina would say! cheers 👍 -)

/etc/network/interfaces pull in the "octopi-network.txt" from the /boot/
```
mapping wlan0
  script /root/bin/map_iface
mapping wlan1
  script /root/bin/map_iface
**source /boot/octopi-network.txt**
```
**configure ur SSID and password in "octopi-network.txt"**

the recommended setting it to leave the `/etc/network/interfaces` as it is and edit the `/etc/dhcpcd.conf` as follows for static IP for both the wLan & Lan 

```
interface eth0
static ip_address=**192.168.10.26**/24
static routers=**192.168.10.1**
static domain_name_servers=**192.168.10.1**

interface wlan0
static ip_address=**192.168.10.25**/24
static routers=**192.168.10.1**
static domain_name_servers=**192.168.10.1**
```
reboot and ur are good to go.. btw another issue is changing the Locale to en_us did not fix the getting "£" instead of "#". or " instead of @

Change keyboard layout to US from default British layout edit the `/etc/default/keyboard`

```
XKBMODEL="pc105"
XKBLAYOUT=**"us"**
XKBVARIANT=""
XKBOPTIONS=""

BACKSPACE="guess"
```
have a great day.. cheers.

orion411, 2017-12-17:
I've tried everything including Raspbian Stretch and can't get my wifi working on wlan0.
This didn't work for me either.
```
iface wlan0-octopi inet dhcp
wpa-ssid "SSID"
wpa-psk "password"
```
I'm still not getting a wlan0 interface.

BillyBlaze, 2017-12-17:
> this is the wrong tracker for this to begin with - what you are having issues with is the OctoPI image, and this is the tracker of the OctoPRINT software running on it. octopi-network.txt is not a part of OctoPrint, it is a part of OctoPi.

Go over to [OctoPi github](https://github.com/guysoft/OctoPi) or use the [Mailinglist](https://groups.google.com/group/octoprint) or the [Google+ Community](https://plus.google.com/communities/102771308349328485741) for such support issues.

Locking this thread.