OctoPrint/OctoPrint #1958: Problem with config.yaml

fbarcenas, 2017-06-06:
#### What were you doing?

Try to insert lines in file config.yaml

Ideally provide exact steps to follow in order to reproduce your problem:

1. I connect with Putty to octoprint 
2. sudo service octoprint stop
3. nano ~/.octoprint/config.yaml
3. I add the lines:

> controls:
>  - name: Parinting Controls
>     type: section
>     children: 
>     - command: M220 S%(speed)s
>       name: Printer Speed
>       type: parametric_command
>       input: 
>       - name: (0-200%)
>         parameter: speed
>         default: 100
>     - command: M221 S%(flow)s
>       name: Flow rate
>       type: parametric_command
>       input: 
>       - name: (0-200%)
>         parameter: flow
>         default: 100
4. Control +X
5. Yes (to save)
6. i do ENTER
7. sudo service octoprint restart
8. I attach capture...
[image]

#### Branch & Commit or Version of OctoPrint

1.3.4

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

Marlin 1.1.0

#### Browser and Version of Browser, Operating System running Browser

Chrome MacOS and Safari

#### Link to octoprint.log

https://pastebin.com/DhnZsipy

foosel, 2017-06-06:
Your yaml is broken:

[image]

note that `name: Parinting Controls` is indented by one space from the `-`, but everything else in that block (`type`, `children`, ...) by two. That is invalid, whitespace *matters* in YAML.

You might want to look into the [Custom Control Editor Plugin](http://plugins.octoprint.org/plugins/customControl/), that allows you adding custom controls without having to edit YAML. Or read through the [YAML Primer in the docs](http://docs.octoprint.org/en/master/configuration/yaml.html). 

Closing because this is not a bug or a feature request but an issue with invalid data.

closed 2017-06-06 by foosel

fbarcenas, 2017-06-06:
So in the website:
https://github.com/foosel/OctoPrint/wiki/Cookbook:-Custom-Controls

is wrong write?
How must be?

foosel, 2017-06-06:
No, on the website there it is correct, but what you put into your `config.yaml` according to what you quoted above does not match what is on this website.

foosel, 2017-06-06:
Your version:

```
controls:
 - name: ...
    type: ...
    children: ...
```

The website's version:

```
controls:
- name: ...
  type: ...
  children: ...
```

Above: name, type, children, ... are NOT on same level. Below: name, type, children, ... ARE on same level.

But really, use the plugin. It will make your live easier.

fbarcenas, 2017-06-06:
yes i see, in the website is ok.
When i do copy/paste the problem happen?

Thanks for this!