OctoPrint/OctoPrint #1959: Reconnecting to server leaves shutdown-dialog open

emente, 2017-06-06:
#### What were you doing?

1.Shutdown Machine that runs OctoPrint via System->Shutdown

Machine shuts down, Shutdown alert in Browser still open
The dialog that confirms a lost data connection to server appears

2.Click reconnect to get UI to initiate server reconnect

UI unghosts and is fully functional, but Shutdown alert still visible.

#### What did you expect to happen?
Modal dialogs disappear after successful reconnect || Modal dialogs disappear after button click

#### What happened instead?
(Already clicked) Shutdown alert still visible, cancel button functional

#### Branch & Commit or Version of OctoPrint
master branch/1.3.4

foosel, 2017-06-07:
I cannot reproduce what I think you are describing, but I'm unsure if I'm looking at the right thing. Could you provide a screenshot or better yet a video showing the behaviour?

emente, 2017-06-07:
Just tried to reproduce the bug and could only after tens of tries. Only thing i can think of is that Raspbian shuts down so very fast that the octoprint process is terminated before the shutdown-dialog is hidden.

As the severity of this bug is very low and it's reproducibility seem to depend on the moon phase... I'm okay with WONTFIX.

foosel, 2017-06-09:
Hm, the hiding of the shutdown dialog happens in the browser without interaction from the server, so how fast the backend shuts down shouldn't influence this.

But if it's tricky to reproduce it definitely sounds like a possible race condition in the frontend code. I'll leave this open and keep an eye out for it.

emente, 2017-08-06:
At first i thought it could have something to do with the fact that i installed OctoPrint on top of a ROS fuerte image (because the Pi+Image are leftovers from my robot who got an waaay faster intel cpu). But last week the Pi died, with a new one (but same sd card) the behaviour is now as expected. 

Thanks for taking time.

Another bug ticket done for ♬

foosel, 2019-11-05:
Closing this since I haven't heard from anyone else running into this for over two years.

closed 2019-11-05 by foosel