OctoPrint/OctoPrint #1960: ERROR - Something crashed inside the serial connection loop

Clotoo, 2017-06-07:
Sorry I have no idea what the printer was doing when this happened, so I cannot provide the usual q/a for bug reporting.

I only found it in some kind of stuck state, and found an error in the logs. This was a one-time thing, as the printer/octoprint usually works fine. Hopefully the stack trace might help figuring it out.

Version: 1.3.0 (master branch)

OS: OctoPi 0.12.0

Printer model : spiderbot
Firmware : Repetier (don't know about the version)

The log : https://gist.github.com/Clotoo/1cc8ef8b4bf9a1af1b3bb4ab67b5eaaa

I'm late for the screenshot but basically, Octoprint UI was simply showing State : `Error (see octoprint.log for details)`. After reconnecting the error was gone and state back to Operational.

foosel, 2017-06-07:
You are still running 1.3.0. Please update to the current stable version. This bug is already fixed there (see #1838)

Clotoo, 2017-06-07:
All right then sorry for the trouble

closed 2017-06-07 by Clotoo