OctoPrint/OctoPrint #1962: MJPG-Streamer flickers on/in iOS 11 Safari

aviatorken89, 2017-06-09:
What were you doing?

Viewing the webcam within Safari on iOS 11

What did you expect to happen and what happened instead?

I expect there to be no flickering, but there was flickering of the webcam stream. Even going to the direct webcam url stream: 'a/webcam/?action=stream' it flickers. The flickering/flashing is almost like a strobe light effect and it makes it hard to look at.

Branch & Commit or Version of OctoPrint

Devel

Operating System running OctoPrint

OctoPi

Browser and Version of Browser, Operating System running Browser

Mobile Safari app on iOS 11

I have read the FAQ.

foosel, 2017-06-09:
The only thing that OctoPrint does here is having an `<img>` tag on the page that points to an MJPG stream, provided by an external webcam server that OctoPrint has absolutely no control at all over. As you already pointed out, the same problem arises when you directly navigate to the webcam server in question (mjpg_streamer in this case), so this is definitely not an issue with OctoPrint and hence nothing I can do anything about at all. 

FWIW, I only have access to a mobile Safari version 10 and that's not showing any flickering.

aviatorken89, 2017-06-09:
Yes, it doesn't do it on less than iOS 11 otherwise I would've noted it. I need to figure out a fix though.

foosel, 2017-06-09:
Sorry, but there's nothing I can do about that. It sounds like a bug on that Safari version, and that is something that Apple has to fix. In the meantime you might have success by using the [SnapStream plugin](http://plugins.octoprint.org/plugins/snapstream/).

I'm closing this since it's not a bug in OctoPrint. There's nothing I can do about this, it's a bug in Apple's software, not anything that I control or can fix.

Comments are still open.

closed 2017-06-09 by foosel

aviatorken89, 2017-06-09:
Why are you assuming it's Apple's fault and that the problem doesn't lie withing mjpg-streamer, which seems abandoned on github looking at how old the included jquery.js version is.

foosel, 2017-06-09:
Because it still works on Safari 10 and everything else I throw at it here and doesn't on your version. When it's the Safari version that changes for breakage to occur, chances are high it's Safari. 

But even if it IS mjpg-streamer - that's also nothing I control. OctoPi (which is NOT OctoPrint mind you, different maintainer, different repo) bundles [this fork of it](https://github.com/jacksonliam/mjpg-streamer) which as far as I know is still maintained (and jquery is not a core component of mjpg-streamer, it's merely a part of the included default web interface, so I can fully understand updating that not taking top priority without necessity). I'm fairly sure though if you open a ticket over there you will hear the same that I told you: Most likely a Safari bug, works flawlessly everywhere else, needs to be fixed by those who maintain Safari.

aviatorken89, 2017-06-09:
Not a Safari bugm it's WebKit; chrome and firefox use WebKit and the problems as I noted are also shown in those browsers as well. I'll see if OctoPi dev has a fix.

foosel, 2017-06-09:
I run Chrome. I run Firefox. Neither show a "flickering stream".

Firefox doesn't use Webkit. Chrome has their own fork, as does Opera. Both differ from Safari's version significantly enough that bugs occurring in Safari don't in Chrome or Opera. Neither show the issue here. If they do for you, I'd make sure it isn't the *webcam* that's actually flickering (e.g. due to a wonky white balance or exposure setting toggling madly back and forth - depending on the nature of the flickering you are seeing that might be the cause too).

And @guysoft won't be able to help you with that problem either, he also doesn't maintain mjpg-streamer (or Safari), he only bundles mjpg-streamer with OctoPi.

I understand that you want this problem fixed, but you are approaching it wrong. If it is showing in other browser as well (which you didn't mention before btw), make sure it isn't the webcam. Test a different one. Test the same one in the same location attached to a different machine. If it used to work and now doesn't, check the cabling, try a different one (things break).

aviatorken89, 2017-06-09:
I edited my post and added the other browsers a couple hours before you replied so I did mention it. They use WebKit on iOS so WebKit is the issue. If you have a Mac with Sierra 10.12, download the Safari Technology Preview and you will see the flickering I bet: https://developer.apple.com/safari/download/

foosel, 2017-06-09:
Apologies, I missed that part in between the multiple edits.

But my point stands. Check if the problem persists on the **desktop** versions of those browsers since otherwise you are merely testing against what is basically Safari with a different skin when it comes to render engine behaviour. If you can't reproduce the issue on a desktop Chrome, it's a bug in Apple's Webkit version they bundle with Safari (again, Webkit in Safari is not the same as Webkit in Chrome, unless in Apple's heavily restricted iOS eco system where they force everyone to use their own Webkit variant). So something Apple has to fix and no-one else can fix but them. If the problem persists on desktop too, I suggest to check your camera.

I don't have a Mac and I don't intend on changing that.

aviatorken89, 2017-06-09:
I have found the root of the problem; it's been around for 3 months that started in Safari tech preview 26. The problem is that for large images WebKit loads the image asynchronously which is an added feature in WebKit but in this case it's the cause of the flickering. Through trial and error I found that images over 453 by 453 in size will be loaded asynchronously. There is a WebKit boollean property: `largeImageAsyncDecodingEnabled` but I am not sure how to set this in css.