OctoPrint/OctoPrint #1963: EventHandlerPlugin does not return payload on FileSelected

identify3d-joe, 2017-06-10:
Reproduce

1. Create test plugin with EventHandlerPlugin
2. Select file
3. payload is empty everytime

#### What did you expect to happen?

payload should be populated with dict

#### What happened instead?

empty payload

#### Did the same happen when running OctoPrint in safe mode?

[Try to reproduce your problem in safe mode. You can find information
on how to enable safe mode in the Contribution Guidelines.]

#### Branch & Commit or Version of OctoPrint

1.3.4.post0.dev0+gfe481e1 (master branch)

#### Operating System running OctoPrint

Raspberry Pi Raspbian

#### Debug

I have actually debugged this problem for you. In the standard.py at line 1172, you have the following function:

`	

      def _payload_for_print_job_event(self, location=None, print_job_file=None, position=None):
		print_job_size = None
		if print_job_file is None:
			with self._selectedFileMutex:
				selected_file = self._selectedFile
				if not selected_file:
					return dict()

				print_job_file = selected_file.get("filename", None)
				print_job_size = selected_file.get("filesize", None)
				location = FileDestinations.SDCARD if selected_file.get("sd", False) else FileDestinations.LOCAL

		if not print_job_file or not print_job_size or not location:
			return dict()

		if location == FileDestinations.SDCARD:
			full_path = print_job_file
			if full_path.startswith("/"):
				full_path = full_path[1:]
			name = path = full_path
			origin = FileDestinations.SDCARD

		else:
			full_path = self._fileManager.path_on_disk(FileDestinations.LOCAL, print_job_file)
			path = self._fileManager.path_in_storage(FileDestinations.LOCAL, print_job_file)
			_, name = self._fileManager.split_path(FileDestinations.LOCAL, path)
			origin = FileDestinations.LOCAL

		result= dict(name=name,
		             path=path,
		             origin=origin,
		             size=print_job_size,

		             # TODO deprecated, remove in 1.4.0
		             file=full_path,
		             filename=name)

		if position is not None:
			result["position"] = position

		return result`

Since print_job_size is set to None at the beginning of the code, then this function will always return an empty dictionary unless print_job_file is empty when passed in. In the case where this function is called from a FileSelected event, that variable is not null, hence we always get an empty dict returned which is incorrect behavior.

I change the line 

`if not print_job_file or not print_job_size or not location:`

to 

`if not print_job_file or not location:`

and the function now operates correctly except that the size field is not set in the returning dict. This is fine for my purposes, but should probably be fixed in your releases.

identify3d-joe, 2017-06-10:
#### What were you doing?

See above

#### What did you expect to happen?

See above

#### What happened instead?

See above

#### Did the same happen when running OctoPrint in safe mode?

Not applicable

#### Branch & Commit or Version of OctoPrint

See Above

#### Operating System running OctoPrint

Doesn't matter, but Raspbian on a Pi

#### Printer model & used firmware incl. version

Doesn't matter, but a Type A Machine

#### Browser and Version of Browser, Operating System running Browser

Really doesn't matter, but Chrome

#### Link to octoprint.log

Not applicable..

#### Link to contents of terminal tab or serial.log

Not applicable

#### Link to contents of Javascript console in the browser

Not applicable

#### Screenshot(s)/video(s) showing the problem:

Not applicable

foosel, 2017-06-12:
This is already fixed with d39c124f35b53a02b80134af28962fe26dc4b1d2 on `maintenance` and `devel` and will be part of 1.3.5.

foosel, 2017-10-17:
1.3.5 was released yesterday.

closed 2017-10-17 by foosel