OctoPrint/OctoPrint #1969: Printer farm...

derfred33, 2017-06-13:
Octoprint has been very useful to run a single printer but what if you have 3 or 4, 10 printers, each one with its own pi and octopi?

Using the api without examples is too much for me and possibly to many more people so why not adding pre made javascript code on the api tab of the configuration page?

This code would be ready to copy/paste with the settings for the machine in question and if one wants to make a webpage with all its machines with general stats like time to completion, temperatures... all that was needed was put those JS pieces of code on an html table cell, div, bootstrap col- whatever.

For local printers why not a simple way to access the info, without api keys.

Salandora, 2017-06-14:
Uhm are you aware of the [docs page](http://docs.octoprint.org)?
There you'll find plenty of examples and explanations, additionally you could use the OctoPrint Client JS library, okay you would need to extract it out of octoprint but that is mostly copy files to another place if I'm not wrong, maybe a little bit of refactoring.
To the API keys, it's also explained how you define them in the config, then just put them into your javascript code and it's done.

derfred33, 2017-06-14:
Very well. So, lets say I have 5 machines, each one with his pi and octoprint instance running, then I have another pi with raspbian, apache and php, this is my web server.

Instead of having 5 tabs open on the web browser I have just one with thumbnails like a gallery, a bootstrap col-md-3 to be specific each one displaying a static image of each machine, when clicked a bootstrap modal opens in full screen and inside an iframe with the actual octoprint page of that machine.

This I have already done and its working very well, apart from the raspbian stubbornly insists on sleeping every 10 minutes or so but this I´ll eventually solve.
Now, the problem is how to get data from the machines, I need the temperatures and the printing remain time.

I can share this code if anyone cares.

I´ve tried to include the two js files mentioned in the docs, jquery I already had, still I commented mine and tried to use the version on octoprint. Failed to load both files.

Of course I didn´t used the templating tags, I replaced by a path to the files none of the octoprint instances. No path worked, couldn´t find a www folder on octoprint, possibly its all managed by the templating system jinja2 or whatever, is unknown to me as is python to my discontent (should I learn python at 40yo? seems interesting, maybe one day...)

If including those js files had worked my next step would be to try the js call from the docs, problem is, at some point it says "your code here or do something with the response..." do what?

Well, maybe its this to get job info:
OctoPrintClient.job.get(opts)

But (opts) ??!! Arguments? Is it Arguments?

What arguments? the machine name? The machine ip? The machine name and the desired field?

This is all very vage you know. 

Maybe to someone used to all of this it isn´t but to someone just wanting to see how a job is doing and never again care about these issues its like a barrier.

derfred33, 2017-08-13:
Well, based on this:
https://github.com/quillford/PrinterView

I´ve done a few modifications:

Html file:
`<!doctype html>

<html lang="en">
    <head>
        <meta charset="utf-8">
        <meta name="viewport" content="width=device-width, initial-scale=1, maximum-scale=1, user-scalable=no">
        <meta name="apple-mobile-web-app-capable" content="yes">
        <meta name="mobile-web-app-capable" content="yes">
        <meta name="apple-mobile-web-app-status-bar-style" content="black">
        <link rel="apple-touch-icon" href="/images/custom_icon.png">
        <link rel="icon" sizes="192x192" href="/images/custom_icon.png">
        <link rel="icon" href="/images/favicon.ico" type="image/x-icon">
        <link rel="stylesheet" href="https://maxcdn.bootstrapcdn.com/bootstrap/3.3.4/css/bootstrap.min.css">
        <link rel="stylesheet" type="text/css" href="style.css">
        <title>AeroFred Print Farm</title>
        <script type="text/javascript" src="http://code.jquery.com/jquery-1.9.1.min.js"></script>
        <script src="https://maxcdn.bootstrapcdn.com/bootstrap/3.3.4/js/bootstrap.min.js"></script>
        <script src="https://github.com/makeusabrew/bootbox/releases/download/v4.4.0/bootbox.min.js"></script>
        <link href="https://gitcdn.github.io/bootstrap-toggle/2.2.0/css/bootstrap2-toggle.min.css" rel="stylesheet">
        <script src="https://gitcdn.github.io/bootstrap-toggle/2.2.0/js/bootstrap2-toggle.min.js"></script>
    </head>

    <body>
    <div class="container-fluid">
      
          <h4>AeroFred<small> Printer Farm</small></h4>
          
      <hr>

      <div id="printerPanels">
  	      <div class="row" id="printerGrid"></div>
      </div>

  		<div class="modal fade" id="addPrinterModal" tabindex="-1" role="dialog" aria-labelledby="addPrinterModal" aria-hidden="true">
  		  <div class="modal-dialog">
  			<div class="modal-content">
  			  <div class="modal-header">
  				<button type="button" class="close" data-dismiss="modal" aria-label="Close"><span aria-hidden="true">&times;</span></button>
  				<h4 class="modal-title" id="addPrinterModalLabel">Add a new Printer</h4>
  			  </div>
  			  <div class="modal-body">
  				<div class="input-group">
  				  <span class="input-group-addon" id="basic-addon1">IP Address</span>
  				  
            <select class="form-control" aria-describedby="basic-addon1" id="newIP">
              <option value="192.168.1.201">Prusa 1</option>
              <option value="192.168.1.211">Prusa 2</option>
              <option value="192.168.1.212">Prusa 3</option>
              <option value="192.168.1.213">Prusa 4</option>
              <option value="192.168.1.214">Prusa 5</option>
              <option value="192.168.1.202">FT-5</option>
            </select>

  				</div>
          <div class="input-group">
            <span class="input-group-addon" id="basic-addon1">Port</span>
            <input type="text" class="form-control" placeholder="" value="80" aria-describedby="basic-addon1" id="newPort">
          </div>
  				</br>
  				<div class="input-group">
  				  <span class="input-group-addon" id="basic-addon1">API key</span>
  				  

            <select class="form-control" aria-describedby="basic-addon1" id="newApikey">
[... 336 lines cut ...]
		}else {
			return 0;
		}
	});
}

function testConnection(ip, apikey){
  	$.ajaxSetup({headers:{"X-Api-Key" : apikey}});
  	$.getJSON("http://"+ip+"/api/version", function(json){
  		if(json.api !== null){
        addPrinter(ip, apikey);
  		}else {
        connectionError(ip, apikey);
  		}
  	})
  	.error(function(){
      connectionError(ip, apikey);
  	});
}
`
The original code made many requests to the server and trow many errors, it would even overload the raspberry pi´s

JS is not my thing at all so doing this already took me a few days and I did basically nothing so if any of you would be kind enough to make this code a little better I would appreciate.

I think the octoprint instances should only be queried if their machines are online and the status should only be verified every few seconds. I´ve managed to put a timer on the update status function  but still sometimes this script goes crazy.

foosel, 2017-08-22:
> I´d like to pull the print history from the history plugin, is this possible? I can´t see anything related to getting info from plugins in the documentation.

That's because plugins define their own APIs, so there's not much I could write into the documentation about them apart from 

> [SimpleApiPlugin] 
> The simple API constructed by OctoPrint for you will be made available under  ``/api/plugin/<plugin identifier>/``

and 

> [BlueprintPlugin]
> The mixin automatically creates a blueprint for you that will be registered under ``/plugin/<plugin identifier>/``.

(both from [here](http://docs.octoprint.org/en/master/plugins/mixins.html)).

In case of the print history plugin, I'd try something like ``plugin/printhistory/history`` (based on [this](https://github.com/imrahil/OctoPrint-PrintHistory/blob/master/octoprint_printhistory/__init__.py#L148) and [this](https://github.com/imrahil/OctoPrint-PrintHistory/blob/master/octoprint_printhistory/static/js/printhistory.js#L116)).

derfred33, 2017-08-22:
Thank you Gina.

I have to see what is a mixin :) I´m not a programmer you know, I´ve just learned a few things as I needed.

zuhra031, 2019-05-05:
@derfred33 - your code does not work anymore :(
Its throwing errors when I try to add a printer it says:

` ReferenceError: numPrinters is not defined index.js:152:7
    addPrinter index.js:152
    testConnection index.js:284 `

I have changed API in html

NotExpectYet, 2019-09-09:
> @derfred33 - your code does not work anymore :(
> Its throwing errors when I try to add a printer it says:
> 
> `ReferenceError: numPrinters is not defined index.js:152:7 addPrinter index.js:152 testConnection index.js:284`
> 
> I have changed API in html

https://git.notexpectedyet.com/NotExpectedYet/OctoFarm

closed 2021-03-31 by LazeMSS