OctoPrint/OctoPrint #1970: Enhancement request: Queue jobs by color/type

CollinChaffin, 2017-06-14:
Requesting a feature to allow queuing jobs by filament type or color.

I would be curious to see a vote, but I know in every 3d print workflow I have ever had what would have made just about every single one more efficient would be the ability to queue all the jobs with the filament loads being the driving factor.  In other words, keep my printer(s) running 24/7 whenever possible with as few filament un/re-loads when I could simply run all the various jobs that need to be in black TPU and on top of that, there is no reason today that we could not have Octo compute how many jobs (based on job settings) will run until a safety filament reload is needed, and re-order my jobs accordingly so the next 20 jobs that need white PLA+ print next.

Since even standalone Cura can fairly accurately estimate (based on defined job filament type) the amount and even approx costs per piece, that someone can't take it one step further and then compute that of my 126 jobs in the queue, I have defined 28 of them as being black PLA so black PLA (all jobs) I can look and see is just shy of 2.6kg at a total running cost of $58 with est overhead, and of all my jobs 21 are white TPU needing 1.9kg at total cost of $62, etc.

Think about it and it's kind of the exact kind of scenario that makes you think of Octoprint, to begin with!

derfred33, 2017-06-14:
I agree that a job focused approach would be helpful and would enrich the system.

-New Job
-Multiple files upload
-Print Job
-Pause Job
-Cancel Job
-Delete Job
-Mark job as done/todo
-Job stats, not only file stats.

znelson710, 2017-06-19:
It would be awesome if OctoPrint could track filament rolls. I know it can see filament used in a job. Just ask for roll info and make a small database and before a print starts ask the user to assign it to a roll and then subtract from the roll that way I can browse my database and pick a roll that will be enough if I have a job that is questionable.