OctoPrint/OctoPrint #1971: [Request] Manual temp adjustments increment from last value vice 0

necro-nemesis, 2017-06-14:
Manual temperature selection defaults to 0 when temperature set by Gcode then increments. It would be more beneficial to be able to tweak a value if manual temperature adjustments started from last temperature set value vice 0. For example if the temperature was currently set to 190 degrees, when using the arrows it would increment/decrement from 190 vice commencing from zero.

halxinate, 2017-07-02:
It's not a feature request, it's a severe bug. Who really wants to "heat" the bed to 0C???

foosel, 2017-07-03:
Everyone who wants to be able to also disable their heater. Which I'd figure are quite a lot of people...

halxinate, 2017-07-03:
Why don't you put –273C there then? (To disable it for sure, physics, you know) :D
Seriously, there is an Off option for that available right there. Why bother with entering 0 from keyboard and then clicking Set when you can click arrow and then Off? 

I'm telling you again: that's the bug. As when yo have something set (e.g. 50C) that value is shown in that Edit field. However, when you click any of the arrows - it drops to 0 instead of modifying the set value. That's what OP is talking about.

foosel, 2017-07-04:
You:
> However, when you click any of the arrows - it drops to 0 instead of modifying the set value. That's what OP is talking about.

Me:
> but instead that if you click on the up/down arrows on the input field in current stable it would start at 0 instead of the current target.

^ I said exactly that. And a fix for that has been in the works for 1.3.5 since OP opened that ticket.

> Why don't you put –273C there then? (To disable it for sure, physics, you know) :D

Because that's not how your printer's firmware works. Your printer's firmware treats "0" as "off", and that's also exactly what the "off" option in that dropdown does as well: it sends `M104 S0` (hotend) or `M140 S0` (bed), so a target of 0°C.

To summarize: It shouldn't do the "start from 0 on arrow button click" thing it does right now and it will no longer do that in 1.3.5. But being able to set a heater to 0 target is possible and will always be possible because it's a valid target temperature that translates to "off" in printer firmware and not being able to set that manually would make a lot of people rightfully angry.

OutsourcedGuru, 2017-07-10:
Actually guys... it's not showing a zero there as the default.  It's trying to show the word "off" and it can only display the first character of that.  And a small oh looks like a zero.

-----------

The input control associated with Target currently can't display more than a single digit/character.  Upon startup, that control is trying to display "off" but it can only manage to show the first letter of that.

If you attempt to enter "123" you can only see the last digit typed.

[image]

In Safari, I attempted to go into the Inspect Element screen.  In this screen you can edit the style, for example, to see the result.  I attempted to change "input-mini" to "input-small"/"input-medium"/"input-large" but this did not change the appearance.  If you delete the "tempInput" class then it will size differently, actually seeing the sizing distinction.  It looks like that tempInput style is unnecessarily overriding the other, forcing it to something like 50x34 pixels.

```
<input type="number" class="input-mini text-right tempInput" data-bind="attr: {placeholder: cleanTemperature(target()) }, value: newTarget, enable: $root.isOperational() &amp;&amp; $root.loginState.isUser(), event: { keyup: function(d, e) {$root.handleEnter(e, 'target', $data);} }" placeholder="off">
```

Within the source, the file in question appears to be `~/OctoPrint/src/octoprint/templates/tabs/temperature.jinja2` and it's down around line 26.

Browser: Safari 10.1

foosel, 2017-07-10:
Whatever that is in your screenshot, it is not the stock ui of OctoPrint. It's either from a modified version of OctoPrint, or a plugin, but in any case not anything I have any control over.

OutsourcedGuru, 2017-07-30:
@foosel  Fair enough.  I created the issue on the OctoPi repository and he suggested that it's your code.  : shrugs :

Kunsi, 2017-07-30:
Please try running OctoPrint in safe mode and check wether the problem persists.

foosel, 2017-10-17:
1.3.5 was released yesterday.

closed 2017-10-17 by foosel