OctoPrint/OctoPrint #1973: GCODE viewer shows incorrect layer number

emgi2008, 2017-06-17:
For layer 1 & 2, the highest two numbers of the print are shown. 
What is actually layer 3 is displayed as number 1. The layer height is shown correctly. 
Example: 
[octo134-layerissue.txt](https://github.com/foosel/OctoPrint/files/1082567/octo134-layerissue.txt)

This behavior was first seen after upgrading from 1..3.2 to 1.3.4. (skipped 1.3.3)

foosel, 2017-06-23:
Can you please provide the requested information or at the very least a file with which to reproduce this? I have nothing to really go on here right now.

emgi2008, 2017-06-23:
Hi Gina,

The requirements of the bug form where simply a bit too much for me. As
such I wasn't counting on a follow-up.
Your personal reply is a very pleasant surprise, very much appreciated!
This shouldn't be very difficult to reproduce: In gcode viewer the first
two layers are shown with a high number, what is actually layer three is
displayed as nr 1. All subsequent layers are numbered in ascending order
but the last one is probably two short of the actual value.
I have a rpi2B+ with v1.3.2 which was updated when offered the option to do
so  on the main page. I believe the problem started after the upgrade to
1.3.4. No tinkering whatsoever with the stock distro and this pi is used
exclusively for 3d printing. Nothing else running on it.

If you need more info, you may also call me and I call you back on my
company mobile. I'm able to speak german if you want although I think your
english is perfect as well.
My number: +31 6 4681 3987

Keep up the good work!
/leo

foosel, 2017-06-23:
The thing is that the issue is not reproducible for me right now with anything I throw at it, hence the request for a file with which to do it ;) What might be happening though is that you have some extrusion happening at what is perceived to be a high z value, meaning that the viewer will sort that accordingly. Could be easily checked if you'd provide a file. If you can't do that publicly for whatever reason, feel free to send it to me by mail to gina -at- octoprint.org.

emgi2008, 2017-06-23:
Hi Gina,

The issue was seen while pringing a file from thingiverse which I sliced
with cura.
Right now I'm out camping but I can send you the file on monday.
This is in no way a critical issue, the print still comes out fine.

Schones Wochenende!
/leo

foosel, 2017-06-23:
Perfect, I'm close to calling it a week myself. Happy camping! :)

foosel, 2017-06-23:
Oh, as a quick follow-up, I need the GCODE file, so the exact sliced result with which you were seeing the issue, not the STL.

emgi2008, 2017-06-26:
Hello Gina,

Here is an example.
Sliced with Cura v2.3.1 and then uploaded to Octoprint.
I'm seeing this behaviour with every file that was printed since the
upgrade.

/Leo

foosel, 2017-06-29:
@emgi2008 your example didn't make it through. You are posting (via email) to [your ticket on the issue tracker](https://github.com/foosel/OctoPrint/issues/1973). Please upload your file somewhere (google drive? dropbox?) and post the link to it. Alternatively send it to me via private mail (gina -at- octoprint.org).

emgi2008, 2017-06-29:
Will do but I'm in Hamburg until tomorrow. So please be patient!
/leo

foosel, 2017-07-05:
I received the file (thanks for that!) but I cannot reproduce what you are describing.

Here's what I did:

  1. uploaded the file
  2. selected it for printing
  3. switched to GCODE viewer tab
  4. scrolled through layers, checked stated z heights

This is what I get directly after opening the viewer:

[image]

Here's layer 24:

[image]

And here's a GIF of scrolling through the layers:

[image]

That's all against OctoPrint 1.3.4. Either I'm misunderstanding your problem or I can't reproduce it.

emgi2008, 2017-07-05:
Hi Gina,

I'm amazed at how much effort you are putting into this.
If you are unable to reproduce it, it probably is a local issue on my
installation. Not sure how that could be but anyway. As it is, it is not
causing any real problems so I'm happy to leave it at this.
Alternatively I can perhaps run some checks if you tell me what to do.

Thnx!
/Leo

foosel, 2017-07-14:
Still all out of ideas of what you could do, sorry. I'll simply leave this open and marked as unreproduced, maybe someone else will report in with the same issue and allow us to triangulate ;)

b-morgan, 2017-09-06:
I read this issue because it was similar to an issue I have with the GCode Viewer (and http://gcode.ws) of showing a blank layer and then being off by one layer. I was able to solve my issue on gcode.ws by turning off GCode Analyzer option "Sort layers by Z". That option (and the others on gcode.ws) don't appear to be tweakable by the user on OctoPrint. Perhaps they should be?

In any case, I believe the sort (or a bug within) could be the source of this issue as well.

foosel, 2017-10-19:
@b-morgan would you be able to share a file with which to reproduce this? I can't debug this without a file with which I can reproduce and I couldn't with the one provided by OP.

b-morgan, 2017-10-19:
I didn't have one saved but the first one I opened shows the problem at layer 7.

[3DBenchy_.25.gcode.zip](https://github.com/foosel/OctoPrint/files/1398736/3DBenchy_.25.gcode.zip)

b-morgan, 2017-10-19:
I think I know what is happening. The start code for the TAZ 6 has a bit of code to wipe the nozzle before doing a bed autolevel. As there are Z values in the wipe code, the gcode viewer sorts it in with the rest of layers in the model.

foosel, 2017-10-19:
Yeah, I just spotted that code snippet as well and was about to comment ^^ Moves around Z10 and Z15, which of course is way above the first layer.

The problem is that those movements are indistinguishable from "actual" print movements. So it's fairly tricky to filter them out or something. And having them sorted under "Layer 0" would also be misleading. Meh.

b-morgan, 2017-10-19:
The only "solution" I can think of is to add a settable parameter that ignores <n> lines at the beginning of the gcode. Since the start code will probably be a constant size for each printer, this is probably a set once and forget it (until it messes up sometime in the future after you forgot about setting it)!

b-morgan, 2017-10-19:
In addition to number of lines to ignore, an option to ignore all lines before a string might also be useful. In my case the string "M117 TAZ 6 Printing..." would work.

foosel, 2017-11-03:
The above commit adds the "sort layers" and "hide empty layers" toggles. I also took a brief look at circumventing this issue in a more general manner by making the layer generation more intelligent, but that produced horrible side effects with other files and is definitely something that will take a while to get right, which I currently can't justify.

foosel, 2019-11-05:
Haven't heard from anyone else running into trouble and the advanced options introduced in 2017 should allow to work around these issues anyhow, so I'm closing this.

closed 2019-11-05 by foosel