OctoPrint/OctoPrint #1975: Deletion of read only timelapse fails silently

ntoff, 2017-06-19:
#### What were you doing?

Trying to break things as usual. I appear to be good at that.

_Ideally provide exact steps to follow in order to reproduce your problem:_

1. Navigate to timelapse directory and set one or more timelapse videos to read only
2. Try to delete one of the timelapses by pressing the garbage can icon.

No error message is displayed notifying users the deletion failed (apart from a java console error which wouldn't normally be visible)

Now try it again with the mass delete feature:

1. (as above)
2. Tick the box next to the timelapse you've set to read only and press "Delete selected"

This time an error message is displayed (although it doesn't say specifically *why* it failed).

[image]

Works on any timelapse file you may wish to "protect" by setting it to read only.

#### What did you expect to happen?

Some sort of error message along the lines of the regular error messages that pop up during failure to capture timelapses, failure to acquire timelapse images, etc

#### What happened instead?

Nothing when using the old method of pressing the garbage can icon.

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.4.0.dev419+g24dc120 (devel branch)

#### Operating System running OctoPrint

windows 7 x64

#### Printer model & used firmware incl. version

none / virtual

#### Browser and Version of Browser, Operating System running Browser

google chrome

#### Link to octoprint.log

pretty sure we know why it can't delete the file :)

#### Link to contents of terminal tab or serial.log

not a serial issue

#### Link to contents of Javascript console in the browser

[image]

#### Screenshot(s)/video(s) showing the problem:

see above

#### Additional Thoughts:

I'm aware that this would be a severely remote chance of a user going and setting a timelapse file as read only but still, isn't it worth putting up a big red "Failed to delete" box just to let the user know OctoPrint *tried* to delete it, and it didn't just silently fail?

I have read the FAQ.

ntoff, 2017-06-19:
possible fix:

\octoprint\src\octoprint\static\js\app\viewmodelstimelapse.js - line 226

	self.removeFile = function(filename) {
        var perform = function() {
            OctoPrint.timelapse.delete(filename)
                .done(function() {
                    self.markedForFileDeletion.remove(filename);
                    self.requestData()
                })
                .fail(function() {
                        var html = "<p>" + gettext("Failed to remove timelapse file");
                        html += pnotifyAdditionalInfo('File: <pre style="overflow: auto">' + filename + '</pre>');
                        new PNotify({
                            title: gettext("Could not remove timelapse"),
                            text: html,
                            type: "error",
                            hide: false
                            });
                    });
        };

        showConfirmationDialog(_.sprintf(gettext("You are about to delete timelapse file \"%(name)s\"."), {name: filename}),
                               perform)
    };

though this doesn't show *why* it failed, is it possible to get the reason easily?

foosel, 2017-06-27:
Thanks for the heads-up, I expanded a bit on your suggestion, it now will show a reason as well:

[image]

[image]

Fixed in `maintenance`, soon `devel` and to be released with 1.3.5.

ScottyMakesStuff, 2017-07-21:
Hi Foosal,
Firstly wanted to say a big thankyou for making Octoprint so awesome!!

I've had a number of timelapses fail over the last few months. Thanks to your update the latest one gave me the message that the tmp file was deleted. My problem is that the previous tmp files weren't deleted and have filled up my 16GB sdcard. I've had a look over SSH and can't find where they're located?

foosel, 2017-07-21:
Timelapse snapshot files are usually stored under `~/.octoprint/timelapse/tmp` or whatever else you have configured under Settings > Folders > Timelapse Temp Folder. But you should also be able to delete them by expanding "Unrendered Timelapses" here (as long as you have the necessary permissions of course):

[image]

ScottyMakesStuff, 2017-07-21:
Thanks Foosel, I missed that link. 
Just cleared 11GB of unrendered timelapses and I'm back in business :)

commit f20185db (2017-06-27): Better handling of timelapse deletion errors

Closes #1975
closed 2017-10-16 by foosel