OctoPrint/OctoPrint #1978: Communication with printer freezes upon sending "N0 M110 N0*125"

CantSayIHave, 2017-06-21:
#### What were you doing?

I recently upgraded my Marlin for my MakerGear M2 due to an upgrade, and now upon connecting,
this occurs:

    Changing monitoring state from 'Connecting' to 'Operational'
    Send: N0 M110 N0*125
    Recv: Error:No Line Number with checksum, Last Line: 0

It then times out to this:

    Communication timeout while idle, trying to trigger response from printer. Configure long running 
    commands or increase communication timeout if that happens regularly on specific commands or 
    long moves.

#### What did you expect to happen?

Not send an `N0` before `M110`, I'd Imagine. The handshake on my Octoprint is `M110 N0`.

#### What happened instead?

Here is everything up to and past the error:

```
Connecting to: COM4
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x41df7d0, open=True>(port='COM4', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: start
Send: N0 M110 N0*125
Recv: echo:Marlin 1.1.3
Recv: 
Recv: echo: Last Updated: 2017-06-04 12:00 | Author: DouglasM
Recv: Compiled: Jun 21 2017
Recv: echo: Free Memory: 5235  PlannerBufferBytes: 1296
Recv: echo:EEPROM version mismatch (EEPROM=V05 Marlin=V38)
Recv: echo:Hardcoded Default Settings Loaded
Recv: echo:  G21    ; Units in mm
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D3.00
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X88.88 Y88.88 Z400.00 E471.50
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z250.00 E200000.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X9000 Y9000 Z30 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P3000.00 R3000.00 T3000.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S0
Recv: echo:PID settings:
Recv: echo:  M301 P25.89 I1.94 D86.53
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z0.00
Recv: ok
Changing monitoring state from 'Connecting' to 'Operational'
Send: N0 M110 N0*125
Recv: Error:No Line Number with checksum, Last Line: 0
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin 1.1.3 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: ok
Send: M21
Recv: ok
Send: M105
Recv: ok T:21.25 /0 B:21.25 /0 @:0 B@:0
```

Of course I can always wait the timeout, but I feel this still shouldn't be happening.

#### Did the same happen when running OctoPrint in safe mode?

Exactly the same thing happens.

#### Branch & Commit or Version of OctoPrint

Version: refs/__gh__/pull/1102/rebase (HEAD -> master branch)

#### Operating System running OctoPrint

Windows 10

#### Printer model & used firmware incl. version

MakerGear M2 on Marlin v1.1.3

#### Browser and Version of Browser, Operating System running Browser

Google Chrome on Windows 10, different computer from server

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.
The Contribution Guidelines tell you where to find that.]

#### Link to contents of terminal tab or serial.log

https://gist.github.com/CantSayIHave/84f886c11a616b4461121ad271daca25

#### Link to contents of Javascript console in the browser

https://gist.github.com/CantSayIHave/dfe551d4043602e6fa3caba5ec4f4f4e

#### Screenshot(s)/video(s) showing the problem:

[Screen snip](http://i.imgur.com/lo3d1Uy.jpg)

I have read the FAQ.

foosel, 2017-06-22:
Sorry to say but OctoPrint is doing everything correctly here. 

Sending the `N0` in front of the `M110` is necessary for compatibility reasons (some firmwares require line numbers with all commands, and some also parse the line number to reset to from the line number the `M110` is sent with instead of the included `N` parameter, hence OctoPrint always sends `M110` with line number and checksum). It shouldn't cause any issues at all when sent like this since it's a valid command (`M110`) with one parameter (`N0`), line numbered and checksummed as necessary (`N0 ... *125`).

And as the initial `ok` right after the start info dump indicates, the second `M110` sent like this also gets read and acknowledged just fine. Sounds more like your firmware is having trouble parsing the second `M110`, either due to a bug in the firmware (you said you did update to a new firmware version, so that might be possible) or due to garbled communication (e.g. it received something like `M110 N0*125` instead of what was sent, or the controller reset introduced garbage bytes in the buffers, have seen that before). 

Might also be that the first `M110` it sends to tickle the printer into action (some printers need that) is the culprit here. What you could try is ticking the box under Settings > Features > Wait for `start` on connect, then reconnect. Maybe that solves the problem for you. 

What you could also try if that doesn't help is experiment with Settings > Serial > Advanced options > Commands that always require a checksum. If you remove `M110` from that it will no longer be forced to be sent with line number and checksum - you might get over the initial bump that way, but would merely patch up a symptom, not its cause. That you run into problems at all hints at some underlying communication problem.

pyr0ball, 2017-06-23:
This issue only recently started happening to me (just started about 2 days ago) with no change in my printer's firmware.

Just adding a "me too" here. Will try to investigate any connection or permissions issues that might be causing it and will report back

foosel, 2017-06-26:
To quote [the contribution guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-do-before-submitting-a-bug-report):

> **Very important:** [...] Also **provide all information** as [described below](#what-should-i-include-in-a-bug-report) and whatever was additionally requested over the course of the ticket even if you "only" add to an existing ticket. The more information available regarding a bug, the higher the chances of reproducing and solving it.

So please: printer model, firmware, logs :)

juergenschubert, 2019-07-31:
Same here for my Prusa MK3s . just started

foosel, 2019-08-01:
> To quote [the contribution guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-do-before-submitting-a-bug-report):
> 
> > **Very important:** [...] Also **provide all information** as [described below](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) and whatever was additionally requested over the course of the ticket even if you "only" add to an existing ticket. The more information available regarding a bug, the higher the chances of reproducing and solving it.
> 
> So please: printer model, firmware, logs :)

foosel, 2019-11-05:
Closing this since OctoPrint is doing everything correct here on first look and repeated pleas for logs from reproductions for a closer look have gone unanswered.

closed 2019-11-05 by foosel

pyr0ball, 2019-11-05:
@foosel my apologies for lack of followup, I ended up reflashing firmware on my end to fix it. No need to re-open