OctoPrint/OctoPrint #1979: [Request] Disable read-only anonymous access

basdp, 2017-06-22:
I would like to prevent access, also for read only (mostly webcam access and downloading my print files) from not logged in users. As far as I know, this is not possible. Is this something that could be implemented via a plugin?

foosel, 2017-06-22:
Nope, it couldn't, but @Salandora is working on #1110 which will allow that. Webcam access however cannot really be restricted by OctoPrint since OctoPrint is not the webcam server, it merely embeds a stream from it and can't do access control for that.

Closing this since it's basically a duplicate of #1110.

closed 2017-06-22 by foosel

Salandora, 2017-06-23:
Just as a side note here, mjpg_streamer is finally able to bind to an address "output_http -l 127.0.0.1"
This way you can restrict the webcam stream, so no one from outside will be able to connect to it.

Sure you'll stop the webcam stream from the control tab too if you don't use a proxy to become access. But as @foosel once mentioned somewhere you can also restrict the access through the nginx