OctoPrint/OctoPrint #1982: Update Error when updating from version 1.3.2 to 1.3.4

jstremmler, 2017-06-23:
Aktualisiere gerade, bitte warten.

++++++++++++++++++++++++++++++++
Aktualisiere OctoPrint auf 1.3.4
++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py" --branch= --force=false "/home/pi/OctoPrint" 1.3.4
Python executable: '/home/pi/oprint/bin/python'
>>> Running: git diff --shortstat
> git diff --shortstat
Error while trying to run command git diff --shortstat
Traceback (most recent call last):
File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 313, in <module>
main()
File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 309, in main
update_source(git_executable, folder, args.target, force=args.force, branch=args.branch)
File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 204, in update_source
if _rescue_changes(git_executable, folder):
File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 180, in _rescue_changes
raise RuntimeError("Could not update, \"git diff\" failed with returncode {}".format(returncode))
RuntimeError: Could not update, "git diff" failed with returncode None

foosel, 2017-06-23:
This looks like something is weird with your git folder (which OctoPrint needs to be in working order for updating). Did you cut power without a proper shutdown first recently or played around with `sudo` where you shouldn't have?

You could try the following. Log into your Pi via SSH (default user/password on OctoPi are pi/raspberry), then:

```
source ~/oprint/bin/activate
cd ~/OctoPrint
git reset --hard
git pull
python setup.py install
sudo octoprint server restart
```

Report back with any error messages you might be getting.

PS: Update problems are usually environment related (corrupted install, network issues, stuff like this) and hence rather suited for the official support mailinglist or G+ community (both linked above) instead of the bug tracker.

jstremmler, 2017-06-23:
Thank you fro the assistance. Followed the advice and it worked, but took a relatively long time to update. Git was corrupted and I had to reinstall it! Then git pull came up with some errror messages but the reinstallation worked. We willsee if I can update to the "next version"....
And I always close OctoPrint with a serrver shutdown command. So I really have no idea what happend....

foosel, 2017-06-23:
Those error messages would have been interesting since they might have hinted at what it was that went wrong ;)

Closing this since it sounds like your problem is solved (for now at least ;)).

closed 2017-06-23 by foosel

jstremmler, 2017-06-24:
Ok, when I restart the server with "sudo octoprint server restart" I get the following error messages:

Traceback (most recent call last):
  File "/usr/local/bin/octoprint", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2707, in <module>
    working_set.require(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 686, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 584, in resolve
    raise DistributionNotFound(req)
pkg_resources.DistributionNotFound: Click>=6.2,<6.3

foosel, 2017-06-26:
> File "/usr/local/bin/octoprint", line 5, in 

This is a regular OctoPi right? If so this looks very very wrong. It should be trying to start `/home/pi/oprint/bin/octoprint`. The fact that it doesn't sounds like `/etc/default/octoprint` might have gotten corrupted or something. It should state 

    # Path to the OctoPrint executable, use this to override the default setting "/usr/bin/octoprint"
    DAEMON=/home/pi/oprint/bin/octoprint

Also, if something like `/usr/loca/bin/octoprint` even does exist it's a case of a `sudo` used where it shouldn't have been. I suggest making sure that the `DAEMON` value in `/etc/default/octoprint` is set correctly as specified above. If it is and your problem persists, something's really messed up with your OctoPi image and you might need to re-install or attempt further repair steps.

In any case, please get in touch on the mailing list or the G+ community instead, which are the official support channels - your problems are 100% with your particular OctoPi instance, not with OctoPrint, so the bug tracker is the wrong place.