OctoPrint/OctoPrint #1983: [Request] Timelapse : More options for snapshot trigger

gege2b, 2017-06-23:
Hi there

I finally set up my mjpeg server and the timelapse stuff and so, I'm in a "timepalse rampage" mode

I came with an idea (witch isn't new) to have an option to trigger the snapshot based on X and/or Y axis position

This idea was (originally ?) done [here, six years ago](http://eclecti.cc/bytes/reprap-controlled-time-lapse-photography)

They have some options : 

- End of Layer : Takes a photograph once at the start of the first layer and then at the end of each layer (it's the actual "on Z change")
- Corner of Layer : Takes a photograph at the minimum Y of each layer
- Least Change between Layers : Minimizes the X and Y distance between photographs on subsequent layers

The last one give a very good result because there is not a lot of movements of the object while printing (look at the videos on the website)

The original code is in python too (originally made for skeinforge) and can be found [on their github](https://github.com/nrpatel/Sprinter/blob/master/photograph.py)

I don't really know how it would be easy to implement these options, given that the original code is made for a  slicer (i.e. to insert relevant Gcode while slicing)

what do you think about that ?

youngiif, 2017-06-24:
I like this idea; very fun watching the flower grow from nothing in your linked video.  I'm not sure if all the events exist at this point (I'm new to octoprint) to handle taking a snapshot at just the right time, but I would lean towards probably.

I think this would be best handled in a plug-in.

In fact there are a lot of ideas I have for a timelapse plug-in that I'm about to start work on, so unless it's decided this request should be added to the core code I will probably try to incorporate your idea.

Some features will require core code changes so it may take a bit longer than expected since I will either need to make those core changes and submit pull requests (after feedback/discussion with octoprint maintainers) or wait for the maintainers to add these changes to the core code or as a last resort,  have my plug-in inject the core code needed (not sure if this is possible in python as I'm just starting to use it but I assume so, also I'm not sure if this would be frowned upon.)

You can see my Brainstorming issue here #1980

gege2b, 2017-07-02:
I just run accros #621 witch I didn't find before posting this request

seems not feasible host-side but some workaround are given

gege2b, 2018-05-15:
I close this feature request because Octolapse plugin can do this almost of the box
The only think needed is a script to add a GCode trigger where needed 

See [here](https://github.com/FormerLurker/Octolapse/issues/109#issuecomment-389130838) for more info

closed 2018-05-15 by gege2b