OctoPrint/OctoPrint #1988: G29 (Bltouch) sent from Octoprint crashes into bed

ckuecker, 2017-06-28:
I recently updated my printer's firmware to Marlin bigfix 1.1.X - [https://github.com/JimBrown/MarlinTarantula](url) 
Now whenever I try to print any GCODE the G29 command crashes into the bed.  I was on an older version of Marlin for the last year and Octoprint worked just fine. 
I can print the same GCODE straight from Simplify3D and the G29 command performs correctly.

foosel, 2017-06-28:
Just for the record: You are stating yourself that you didn't change anything about OctoPrint but changed your firmware. So most likely your issue is also caused by your firmware, not OctoPrint.

OctoPrint doesn't do any special handling of `G29` or move your bed at will. When you tell it to send `G29` either manually or through a printed GCODE file it will send this just as it to the printer - and only then! It's the job of your printer's firmware to ensure those commands are executed properly and without doing any harm to your printer. See also ["OctoPrint is telling my printer to do wrong things" in the FAQ](https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-is-telling-my-printer-to-do-wrong-things).

ckuecker, 2017-06-28:
Correct.  The only thing that has changed is the firmware.  However, I can print the GCODE just fine via USB from Simplify 3D.  There is some difference between the way OctoPrint and Simplify 3D handles the commands.

fiveangle, 2017-06-28:
G29 works fine on my system also running ~3 day old rcbugfix1.1.1

foosel, 2017-06-29:
OctoPrint doesn't handle that command in any way. It just sends it 1:1 to your printer when you enter it either into the terminal or have it in a GCODE file you print.

The as of yet missing logs (especially the `serial.log`) will show what is being sent to your printer. Compare that to what S3D produces. Might also be some EEPROM settings you have S3D instructed to send to your printer on connect but not OctoPrint. But it's definitely not a bug in OctoPrint.

Grogyan, 2017-06-29:
@ckueker if you change something with the MarlinFirmware, like updating it, and it isn't responding to sliced commands generated from S3D,  please raise that as an issue on the MarlinFirmware Github

foosel, 2017-07-24:
Closing this since OP sadly still didn't provide any logs that might allow to help them and the reported issue is not reproducible.

closed 2017-07-24 by foosel