OctoPrint/OctoPrint #1992: [Request] Reverse second extruder option.

Pueltamika, 2017-07-03:
Hi,

I recently bought a BCN3D Sigma R17, a printer with a independent dual extruder.
For some reason they decided to mirror the second extruder, changing the direction of the steppermotor.
Printing with an SD card using the printers own software works, but the direction of the filament is reversed when using Octoprint.
The obvious option is to reverse the wiring of the printer, but the printer will lose it's ability to load and unload it's filament, or cabilbrate the hotends.

Could it be possible to have a checkbox in the printer settings of octoprint to reverse the direction of the second extruder?

Yours sincerely,

Pueltamika

Attachments: Left and right extruder.

[image]
[image]

lf-, 2017-07-25:
Can you fix this in the slicer?

foosel, 2017-07-26:
If the same GCODE from SD works but streamed from OctoPrint doesn't, you're most likely missing something in your start GCODE that would tell your printer's firmware to reverse the coordinates it receives (some EEPROM value maybe?) that it does add automatically when printing from its internal SD.

While something like reversing every single extrusion line could be done through OctoPrint, it would add *significant* processing overhead, increasing the likelihood of stuttering and negative quality impact severely. There is a reason OctoPrint keeps its hands off of GCODE sent through it and just relies on it being fitting for your printer (as the corresponding configuration options say, all the things like motor direction and such in the printer profile are only used for stuff like the Control panel or making sure the GCODE analysis and viewer behave - it's NOT for modifying files streamed through OctoPrint on the fly!).

Instead you should try to figure out what your printer's firmware does differently when printing from SD vs. when it receives GCODE move commands through the serial connection, and add whatever it's missing either to your slicer or OctoPrint's own start GCODE script.

Closing this because it's simply entirely impractical to allow inverting certain coordinates for files streamed through OctoPrint. It might be possible to write a plugin to solve this, but I highly recommend against it due to the aforementioned performance reasons.

closed 2017-07-26 by foosel