OctoPrint/OctoPrint #1994: [Request] Printer name in page title

MrSourBalls, 2017-07-04:
I am currently running 5 printers on Octopi, each on its seperate RPI3

I would love to be able to see in the blink of an eye what printer i am looking at in my browser, as i plan to expand to +- 10 printers. 

Would that by any chance be possible?

nophead, 2017-07-04:
Settings / Appearance / Title

foosel, 2017-07-04:
^ what @nophead said. Closing.

closed 2017-07-04 by foosel

JoshMcCullough, 2019-02-28:
Seems like what is really needed is variable replacement in the title property. I wouldn't want to hard-code the machine name in Settings > Appearance > Title because if I have multiple machines attached, I'd want the title bar to show the _currently connected_ machine's name. Maybe just a `%connected_printer_name%`, or some less verbose option for replacement in Settings > Appearance > Title?