OctoPrint/OctoPrint #1995: Access to infos from GcodeViewer over REST API

derboehsevincent, 2017-07-06:
Hi, is there a way to get infos like current layer/total layers over the REST API? Yes, I that know that DisplayZ Plugin and I have no idea why it is referenced all the time. (Displaying the Z height in mm has nothing to do with current layers/total layers from my POV)

ntoff, 2017-07-06:
This is a bug tracker, not a help desk (as clearly stated in all the nice helpful information you were supposed to read through before posting this "issue").

http://docs.octoprint.org/en/master/api/job.html details how to acquire the information available about the current job.

foosel, 2017-07-11:
And also http://docs.octoprint.org/en/master/api/push.html.

Otherwise closing because it's not a bug report, feature request or brainstorming ticket and better suited for the established support channels as already mentioned by @ntoff

closed 2017-07-11 by foosel