OctoPrint/OctoPrint #1996: Unable to preview Prusa MultiMaterial Gcode.

ryanneufeld, 2017-07-06:
#### What were you doing?

I uploaded a gcode file that came from the output of the Prusa MultiMaterial PostProcessor.

#### What did you expect to happen?

I expected the gcode viewer to render the gcode preview.

#### What happened instead?

The gcode viewer did not display the preview.

#### Did the same happen when running OctoPrint in safe mode?

[Try to reproduce your problem in safe mode. You can find information
on how to enable safe mode in the Contribution Guidelines.]

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.4 (rc/maintenance branch)

#### Operating System running OctoPrint

OctoPi 0.11.0

#### Printer model & used firmware incl. version

Prusa i3 Mk2 3.0.12MM

#### Browser and Version of Browser, Operating System running Browser

Chrome latest.

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

The only output in the logs is 

    #### Layer number: 0 , max layer: 1

#### Screenshot(s)/video(s) showing the problem:

[image]

foosel, 2017-07-07:
Please provide an example file with which to reproduce this. Either upload it somewhere and link it here or send it to gina -at- octoprint.org with "Issue 1996" as subject.

I can't debug this without an example to reproduce :)

foosel, 2017-07-20:
Got a file provided by someone else (thanks!), could reproduce. Turns out Slic3r Prusa Edition produces those outdated `\r` line endings for MM GCODE files for some reason, and the GCODE viewer so far couldn't recognize those as line endings. Hence all lines separated thus ended up as one line and stuff got parsed wrongly.

Fixed with the above commit on `maintenance`, soon `devel`, to be released with 1.3.5.

commit 7e32e5be (2017-07-20): GCODE viewer: Support \r, \n and \r\n line endings

So far we only supported \n and \r\n, leading to problems with MM
GCODE files generated by Slic3r Prusa Edition which apparently
produces \r line endings for some reason.

Closes #1996
closed 2017-10-16 by foosel