OctoPrint/OctoPrint #1997: Webserver and communication to pi stop, print continues

gspitman, 2017-07-08:
While printing, and watching the webcam, the webserver stops.  I'm unable to SSH or ping the pi.

The print continues, and will complete but I cannot communicate with the server/printer.

ntoff, 2017-07-09:
>I'm unable to SSH or ping the pi.

Nothing to do with octoprint then. Octoprint only handles the connection to your printer, network connectivity is handled by the underlying operating system that octoprint is running on.

foosel, 2017-07-11:
Like @ntoff mentioned that is not an OctoPrint issue but probably rather an issue with either your router or your Pi's network interface. Please get in touch on the Mailinglist or the G+ community (both linked by the bot above) which a better suited place to get help on issues like that :)

Thanks!

closed 2017-07-11 by foosel