OctoPrint/OctoPrint #1998: responsive interface

leobel96, 2017-07-09:
I start complimentanting for the very good work. I want to suggest to add a responsive UI for the use with smartphones with bigger buttons and simpler commands so that I haven't to zoom in and zoom out every time I want to click something or read words. Thank you.

closed 2017-07-10 by leobel96

BillyBlaze, 2017-07-10:
@leobel96 take a look at plugins such as [TouchUI](http://plugins.octoprint.org/plugins/touchui/) or [ScreenSquish](http://plugins.octoprint.org/plugins/screensquish/)

leobel96, 2017-07-10:
@BillyBlaze TouchUI is for the use of a screen connected directly and physically to the octoprint device, I have already tried it. On the contrary ScreenSquish seems to be what I was looking for, thank you so much. However I think that the responsiveness is a very important feature and should be included directly inside octoprint without the use of thirdy party plugins.

BillyBlaze, 2017-07-10:
Actually, as the creator of TouchUI, It has been made for mobile and tablets. Later I added support for connected touchscreen devices. TouchUI will auto-detect every touch device and makes it responsive.

leobel96, 2017-07-10:
Yesterday I installed the plugin, I rebooted system and then I connected using my smartphone but nothing happened, so I concluded that the plugin only affectes connected displays. I'll try again. Thank you.

leobel96, 2017-07-10:
@BillyBlaze I have now tried to reinstall it and now it works and I can use it from my smartphone's browser. Thank you!!!

BillyBlaze, 2017-07-10:
Your welcome & I'm glad it works 👍 Enjoy