OctoPrint/OctoPrint #1999: [Request] Responsive UI

leobel96, 2017-07-10:
I start complimentanting for the very good work. I want to suggest to add a responsive UI for the use with smartphones with bigger buttons and simpler commands so that I haven't to zoom in and zoom out every time I want to click something or read words. Thank you.

ntoff, 2017-07-10:
There is a plugin designed specifically for that http://plugins.octoprint.org/plugins/touchui/

foosel, 2017-07-10:
Just to refer the earlier ticket made by @leobel96 on that matter: #1998.

Now to the matter at hand: I agree that a responsive UI built into OctoPrint from the get-go would be nice to have. It's not trivial however - in order to do this properly the UI would need to be rebuild from the ground up, and with newer libraries to boot (we are still stuck at the now ancient Bootstrap 2). And that in turn has huge implications when it comes to backwards compatibility for plugins that depend on the current UI. 

Hence not a low hanging fruit by any means.

I added UiPlugins to allow for something like a smooth transition period for such a case - we could build a new UI and users could decide which one they want to use, with the caveat that at the beginning most UI facing plugins would probably not be compatible and need some love from their authors first. It will however not happen before the work on 1.5.0 begins - the roadmap for 1.4.0 is already too full.

I was also absolutely sure that there already was a ticket on this somewhere in the bug tracker (making this a duplicate), but right now I can't seem to find it myself O_O

leobel96, 2017-07-10:
Perfect, thank you. I'll wait for it :)

leobel96, 2017-07-10:
Sorry I have another question: is there a way to change the position of navbar icons? I mean only their position inside the navbar (e.g. controls tab as the first icon, webcam tab as the second one and so on) and to hide tabs that I don't use?

Salandora, 2017-07-10:
http://docs.octoprint.org/en/master/configuration/config_yaml.html#appearance

This should partially help you. There is described how to reorder the tabs and how to disable them.
What you cant to is split controls and webcam into different tabs.

leobel96, 2017-07-10:
@Salandora thank you, I'll try It.

closed 2017-08-10 by leobel96