OctoPrint/OctoPrint #2005: [REQUEST] BLTouch + ATX PSU control with a single RAMPS?

Oscar7601, 2017-07-12:
Thank you very much for the job you did explaining how to control an ATX PSU with Octoprint and a RAMPS board.

I have a question however. I would like to use a BLTouch auto-leveling sensor _and_ control my ATX PSU with one single RAMPS board.

The thing is that in your tutorial you show you have taken the jumper off the pins next to the reset button, but I beleive to connect the BLTouch sensor to the RAMPS board I have to put the jumper on the VCC and +5V pins.

I am absolutely new to both GitHub and this kind of electronics so I apologize if my question is stupid.
Thank you in advance for reading this.

foosel, 2017-07-13:
I have no idea what tutorial you are talking about, sorry. If you are referring to something on the Wiki ([this one maybe?](https://github.com/foosel/OctoPrint/wiki/Control-your-printer's-ATX-PSU-through-a-RAMPS-board-using-OctoPrint)), that was created by some user just like you (the linked one was last edited by @AndrewBCN), not me, so asking me for clarification here won't do you any good I fear ;)

In general, please use the [Mailinglist](https://groups.google.com/group/octoprint) or the [G+ Community](https://plus.google.com/communities/102771308349328485741) for any kind of support request or asking questions and such. This is a bug and feature request tracker, and having it used as a support forum makes development that much harder :)

I'm closing this now, comments are still open though so if @AndrewBCN or someone else who's qualified to answer your question should read this, replying is possible.

closed 2017-07-13 by foosel

Oscar7601, 2017-07-13:
Oops I'm really really sorry @foosel  for not using this section properly.
I'll know better next time!

jneilliii, 2017-07-15:
Check Plugin Manager for OctoPrint-BLTouch.  It adds BLTouch buttons to the Control tab and I recommend the PSU Control plugin for easily controlling the PSU from OctoPrint.