OctoPrint/OctoPrint #2006: Request: Shorten long startup due to implementing SystemMenu Printer Power

louvanna, 2017-07-13:
After implementing the system menu for printer power-up and power-down on OctoPrint 1.34, the start-up time is long.  A relay has been added to the printer power supply which is defaulted to 'off'.  The page eventually loads with the notification for "Communications Error". I believe this to be technically correct, the printer is in-fact, powered down. Otherwise, all is well with the OctoPrint after the power-up is issued, and a manual <Connect> is clicked.

I believe this is due to the OctoPrint trying to connect to initially powered-down rambo-board, of a RostockMax V3 printer. 

Is there a way to avoid the connect to printer attempt in the initial invocation, or a similar work around ?

DETAILS of machine 
-------------------
Version 1.3.4 (master branch)
pi@octopi:~ $ cat /etc/octopi_version
0.13.0
pi@octopi:/var/log $ uname -a
Linux octopi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux

Printer: RostockMax v3
2017-07-13 01:48:28,795 - octoprint.util.comm - INFO - Printer reports firmware name "Repetier_0.92.2"

Desktop Browser: Firefox 54.0 (64-bit) on Linux Xbuntu 14.2
$ uname -a
Linux d690 3.13.0-123-generic #172-Ubuntu SMP Mon Jun 26 18:04:35 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

----------
2017-07-13 01:42:28,843 - octoprint.plugin.core - INFO - 11 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/announcements
|  Automatic Shutdown (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_automaticshutdown
|  Backlash autocalibration (0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_autocalibration
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/discovery
|  Email Notifier (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_emailnotifier
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate
|  TemperatureFailsafe (0.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_TemperatureFailsafe
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/virtual_printer

AND the power-up command near the end of the current log.

2017-07-13 01:47:41,820 - octoprint.server.api.system - INFO - Performing command for custom:printer on: /home/pi/scripts/power on

foosel, 2017-07-13:
> Is there a way to avoid the connect to printer attempt in the initial invocation

Yes, uncheck that checkbox which you probably checked at some point, then reconnect once manually:

[image]

Stock OctoPrint doesn't auto-connect on startup, you have to tell it to.

louvanna, 2017-07-14:
Guess, the request is unnecessary since the requested functionality already exists.
thanks.

closed 2017-07-14 by louvanna