OctoPrint/OctoPrint #2007: Temperature Graph not correct in dual extrusion setup with one disconnected extruder

aurror, 2017-07-13:
#### What were you doing?

Having a dual extrusion printer with currently only the second extruder equipped.
Heating extruder 2 up.

note: Printing works fine, only the UI shows something not correct.

#### What did you expect to happen?
That the temperature graph shows both extruders temperature correct. 
Nothing/off for extruder 1, and the current value/target value for extruder 2.

#### What happened instead?
When the currently selected extruder is ex1, the graph will show the target temperature correctly, but the current temperature of extruder 2 is shown as Off, even though the printer reports them correctly, for example:
Shown in UI: Target2: 30 current2: Off
Output:
T:-46.76 /0 B:24.31 /0 B@:0 @:0 T0:-46.76 /0 @0:0 T1:33.58 /30 @1:0

When the currently selected extruder is ex2, the graph copies the temperature value from the second extruder into the first. For example, if I set the temperature to 50°C, both extruder have the exact same values displayed in current temperature and target temperature. 
The temperature graph copied the temperature from extruder 2 into extruder 1, so the value for the first extruder was shown wrongly.
example: 
shown in UI: Target1: 50 Target2: 50 current1: 50.45 current2: 50.45
output:
Send: N231 M105*39
Recv: ok 231
Recv: T:50.45 /50 B:25.58 /0 B@:0 @:0 T0:-55.00 /0 @0:0 T1:50.45 /50 @1:0

 Physically the temperatures were correct, they were only shown wrong on the temperature graph.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Branch & Commit or Version of OctoPrint
OctoPrint 1.3.4 (master branch)

#### Operating System running OctoPrint
Linux - UDOObuntu 2.1.2

#### Printer model & used firmware incl. version
PRINTit 3020 with Repetier Firmware 1.0.0dev

#### Browser and Version of Browser, Operating System running Browser
Google Chrome  59.0.3071.115 

#### Link to octoprint.log
https://gist.github.com/aurror/9f6a3b4c0a08f5c6ab6091898160a7ea

#### Link to contents of terminal tab or serial.log
https://gist.github.com/aurror/c7d527e15d99d8367db11c6d76265f5f

#### Link to contents of Javascript console in the browser
https://gist.github.com/aurror/726504e1ba62d0786af733c87ad50eaf

#### Screenshot(s)/video(s) showing the problem:
-
I have read the FAQ. -Yes

aurror, 2017-07-14:
Thanks for the quick fix!
My printer uses repetier-firmware instead of smoothieware, i don't know if they are similar.
At first, the temperature of the currently selected extruder for extruding is reported, and later on the temperatures for both T0 and T1:
**T**:50.45 /50 B:25.58 /0 B@:0 @:0 **T0**:-55.00 /0 @0:0 **T1**:50.45 /50 @1:0

foosel, 2017-07-14:
Everyone does it slightly different (because there is no standard, hooray!) and the code to try to make sense of the M105 response line across all firmware variants is hence [a bit complex](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/util/comm.py#L2914-L3000). You ran into something it didn't yet correctly cover by introducing negative temperatures into the mix which led to `T0` not being detected as a temperature value at all.

commit cf053ddb (2017-07-14): Temperatures may in fact be negative

E.g. in case of a disconnected thermistor, or when it is really
really cold. Make sure our parsing doesn't get confuse by this and
thinks that

  ok T:150.0/210.0 T0:-55.7/0 T1:150.0/210.0

is actually only T and T1 being reported by the firmware, causing
stuff to be wrongly canonicalized thanks to T/T1 being Smoothie's way
of reporting dual extruder setups.

Fixes #2007
closed 2017-10-16 by foosel