OctoPrint/OctoPrint #2009: [Request] delete more then one file

jon012198, 2017-07-16:
to delete multiply files on sd card at once

ntoff, 2017-07-17:
I believe the filemanager plugin allows selecting multiple files by checkbox and mass deletion https://github.com/Salandora/OctoPrint-FileManager

jon012198, 2017-07-17:
@ntoff  it dose but not a select all button just have to ctrl click them  to delete them

ntoff, 2017-07-17:
Ahh I see. A "select all / select all successful" option might be handy, a bit like the new timelapse page.

I usually just log in via ssh and rm * everything in the uploads folder, probably not the best solution.

jon012198, 2017-07-17:
i did not thing about doing a ssh

ntoff, 2017-07-18:
You can actually make it run shell commands via the system menu, previously I had a "Clean up timelapse folder" entry that did rm *.mpg (or is it mpeg?) up in the system menu, up where the start / restart / shutdown stuff is in the navbar.

I do like the idea of a "remove all successful prints" option though, that's something I don't think would be so easy via ssh and I'm not entirely sure removing all the files via rm * is safe, @foosel ? Does that mess with the metadata if you manually remove uploaded gcode via ssh + rm?

foosel, 2017-07-26:
@ntoff Only just now saw that question, sorry. Removing is fine. I might even have built-in functionality to automatically remove metadata entries for files that are missing. Not entirely sure though.

I'd strongly recommend against deleting stuff though while a print is running, to reduce the chances of accidentally nuking the file you are currently printing ;)