OctoPrint/OctoPrint #2012: CLI softwareupdate init_client not defined

kantlivelong, 2017-07-17:
#### What were you doing?

/home/pi/oprint/bin/python /home/pi/oprint/bin/octoprint plugins softwareupdate:check

#### What did you expect to happen?

Check for updates

#### What happened instead?

Python error

`  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/cli.py", line 40, in check_command
    client.init_client(cli_group.settings)
AttributeError: 'module' object has no attribute 'init_client'
`

#### Did the same happen when running OctoPrint in safe mode?

N/A

#### Branch & Commit or Version of OctoPrint

1.3.4 (master). Checked devel source and init_client is not defined

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

N/A

#### Browser and Version of Browser, Operating System running Browser

N/A

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

I have read the FAQ.

kantlivelong, 2017-07-17:
Upon further investigation it appears that init_client was removed from octoprint_client in commit d72c7c144e2d66cdb3fb09f54ea4149146bb212d

foosel, 2017-07-18:
Fixed on `maintenance`, soon `devel`, will be rolled out with 1.3.5. Thanks!

kantlivelong, 2017-07-20:
Confirmed fixed.

commit 5db9acb3 (2017-07-18): SWU: Fix CLI commands

Client refactoring in d72c7c14 caused issues here.

Closes #2012
closed 2017-07-20 by kantlivelong