OctoPrint/OctoPrint #2013: watcg=hed folder not working with .gcode files

Tinchus2009, 2017-07-17:
----

#### What were you doing?

I set u the watched folder. Looks ok, but when saving a .gcode file into the mapped watched folder, octoprint never sees that file. I have checked "activily poll..." and also doesnt work. 
In contrast, if I save in the watched folder a .stl file, the files is perfectly seen by octoprint

Ideally provide exact steps to follow in order to reproduce your problem:

1. ... Instlled samba and created the watched folder folloing these steps: https://www.youtube.com/watch?v=RQ3ufPo95lg
2. ... Haveing mapped the watched folder in a windows machine, I slice and save a .gcode file in the watched folder
3. ... Return to octoprin and the files is never seen. 
Doing the same but saveing a .stl file works perfectly

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

YES

#### Branch & Commit or Version of OctoPrint

Version 1.3.4 (master branch)

#### Operating System running OctoPrint

[OctoPi, 0.14.0

#### Printer model & used firmware incl. version

Prus I3

#### Browser and Version of Browser, Operating System running Browser

win 10  Firefox 54.0.1 (32-bits

#### Link to octoprint.log

[octoprint.zip](https://github.com/foosel/OctoPrint/files/1153900/octoprint.zip)

#### Link to contents of terminal tab or serial.log

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log (which the Contribution Guidelines tell you where to find).]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.

The Contribution Guidelines tell you where to find that.]

#### Screenshot(s)/video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

foosel, 2017-07-18:
Have you also tested without samba shares in the mix, by simply copying something to `~/.octoprint/watched` on the command line, e.g. `cp ~/.octoprint/uploads/some_file.gcode ~/.octoprint/watched/some_file_watched.gcode`? I cannot reproduce the issues you are seeing with the plain watched folder right now, works just fine here.

foosel, 2017-09-28:
Closing due to lack of reply. Feel free to request a reopen once you can provide the requested information.

Leemonoid, 2017-12-17:
foosel, I can verify that I have the same issue.  I use WinScp to copy a file from my windows server to the "watched" folder but it is never picked up by OctoPrint.  If I SSH to the Raspberry Pi and make a copy of the file, the copied file is discovered by OctoPrint immediately. Both files had exactly the same permissions in the folder.  The EarbudHolder.gcode file below is ignored by OctoPrint.

-rw-r--r-- 1 pi pi 5777357 Dec 17 13:22 EarbudHolderCopy.gcode
-rw-r--r-- 1 pi pi 5777357 Dec 17 13:08 EarbudHolder.gcode

foosel, 2017-12-18:
@Leemonoid please provide `octoprint.log` and information on what version of OctoPrint and if applicable OctoPi you are running.

Leemonoid, 2017-12-18:
OctoPrint version 1.3.6
OctoPi version 0.13.0

I opened WinSCP on my Windows 10 computer and transferred file "TestGcodeFromWin10.gcode" from my Windows machine to the "watched" folder.  The file was not processed by OctoPrint.

I SSH'ed into the Raspberry Pi and copied the same file, named "TestGcodeCopiedFromRPi.gcode", into the watched folder.  It was processed immediately by OctoPrint.

My server log is attached.  Thank you so much for this wonderful program!

[Leemonoid_octoprint.log](https://github.com/foosel/OctoPrint/files/1568137/Leemonoid_octoprint.log)

foosel, 2017-12-18:
I tried the same thing and I cannot reproduce this right now against 1.3.6 and OctoPi 0.14 (no 0.13 on hand right now). Even called the file the same. This is odd, and nothing in your log hints at any kind of problem.

I'll give this another go against 0.13, however, I'm currently wrapping things up for 2017 so this will have to wait until January.

Leemonoid, 2017-12-18:
Happy Holidays to you!  I'm happy to assist further with debugging in the new year.

I loaded OctoPi 0.14.0 and then upgraded to OctoPrint 1.3.6,
The same problem persists with a fresh install of OctoPi 14.0

foosel, 2019-11-05:
@Leemonoid Are you still seeing this?

Leemonoid, 2019-11-05:
@foosel , yes I saw your message

Leemonoid, 2019-11-08:
@foosel The problem seems to be resolved (Octopi 15, with the latest version of Octoprint.) I used Winscp to open an SSH connection to the pi, and dragged a gcode file from my Windows 10 desktop into the "watched" folder. It was ingested by  Octoprint as expected, and the file was deleted from "watched."

foosel, 2019-11-08:
Great. Tentatively closing this then. Should it say hello again, feel free to ping me!

closed 2019-11-08 by foosel